greiman/SSD1306Ascii #2: Great library, but no <newline> using println

gerardwr, 2016-01-22:
Hello Bill,

I have just discovered your library and I REALLY like it :-)

It's compact, yet rich in functions for "normal" text. Thanks for sharing!

But..... I have an "issue".

When I upload your example AvrI2c128x64 (Arduino 1.6.7) to an Arduino-Nano with a "noname" SSD1306 display (which looks identical to yours) the PRINTLN statements function like they were PRINT statements. So, following  the output from a PRINTLN statement the following PRINT or PRINTLN statement is NOT printed on the next line, but immediately after the 1st PRINTLN.

When I ad \n to the String to be printed the following line DOES print the next line. When I WRITE("\n"); also the next output appear on the next line.I'm not sure if this makes any sense, here's an example.

```
  oled.println("Hello world!");
  oled.println("A long line may be truncated");
```
results in
```
  Hello World!A long line may be tru.....
```
But
```
  oled.println("Hello world!\n");
  oled.println("A long line may be truncated\n");
```
results in
```
  Hello World!
  A long line may be tru.....
```

In a JPG of the testoutput on your display is seems to work as expected.

Do you have a clue why I could have this issue?

Thanks.

gerardwr, 2016-01-22:
Just for fun I upload the standard sketch (without the extra \n) using Arduino 1.6.4.

Works like expected, so after every PRINTLN the next output is printed on a new line.

So it seems my issue is caused by Arduino 1.6.7 @#$%^&*()

greiman, 2016-01-22:
Please try 1.6.7 and let me know if it is fixed.

closed 2016-01-22 by greiman

gerardwr, 2016-01-22:
Yes, works OK now in Arduino 1.6.7, and is still OK in 1.6.4.

Thanks for the quick response, brilliant!

hood2309, 2020-09-28:
hi, i just wanted to know if there is a way the longer texts can go to the next line. while running your code, if the text is longer than certain characters it doesn't show those texts