greiman/SSD1306Ascii #3: Request : Add scroll feature to PRINTLN

gerardwr, 2016-01-24:
Hi,

I use your library mainly as a replacement for the Arduino Serial Monitor, using the print and println statements.

In the current version println fills the screen up to the last line on the display, and subsequent println statements also outputs on the last line of the display.

For me the wanted behavior would be the display to scroll up the content of the display when using println on the bottom line of the display.

I added some "proof of concept" code to the function "write" in the file Ssd1306Ascii.cpp (see below). This works fine, but has the flaw that the screen starts scrolling immediately after EVERY println statement, instead of scrolling ONLY  at a println on the last line.

Perhaps you would consider to add perfectly working scroll feature, si I can throw my own code away :-)

Thanks.
===============================
Proof of concept code for println scroll feature 

I replaced
```
    if (ch == '\n') {
      setCursor(0, m_row + m_magFactor*nr);
      return 1;
    }
```
with
```
  if (ch < first || ch >= (first + count)) {
    if (ch == '\n') {
	  m_row = m_row + m_magFactor*nr;
      // when m_row > 7 reset to 0
      if (m_row > 7) m_row=0;
      setCursor(0, m_row);
	  // clear the line from old content
	  clearToEOL();
	  // scroll the entire screen 8 pixels up
	  ssd1306WriteCmd(SSD1306_SETDISPLAYOFFSET);
	  ssd1306WriteCmd( (m_row +1) * 8);
	  return 1;
    }
```

greiman, 2016-01-24:
Sounds like a nice feature.  I will look into adding this soon.

gerardwr, 2016-01-24:
Excellent :-)

I'll stop hacking around your library, and use my "flawed" code for now, awaiting an update from you.

Thanks.

greiman, 2016-01-26:
I probably will only support scrolling on displays that have a display height of 64.  

I tried a 32 high display but it requires a lot of mods.  A 32 high display currently uses the first half of the controller memory.  You need to use all of the memory for scrolling to work.

Also you really only get three lines on a 32 high display which is not too useful.

I think I will have a call to enable/disable scrolling.  I will only start scrolling when the first new line would go beyond the end of the display.

gerardwr, 2016-01-26:
Considering the fact that I have only 64-line displays I agree with the announced limitation ;-)

I can understand the extra work that's required for a 32 line display, and I agree that a scroll feature for 3 text lines is not very useful. I guess most people will choose the 64-line display.

Yes, scrolling can start when a newline occurs on the bottom text line of the display.

I'm not sure for the need of an enable/disable scroll. For me, an always enabled scroll would be fine. A user that does not want to scroll, can alway use setRow and print to prevent scrolling.

Do let me know if you would like me to test something.

greiman, 2016-01-27:
I have added scrolling.  Please try it.

You should look at the ScrollingSPI and ScrollingWire examples.

You may want to edit this section of SSD1306Ascii.h and set INCLUDE_SCROLLING to two.

```
/** Set Scrolling mode for new line.
 *
 * If INCLUDE_SCROLLING is defined to be zero, new line will not scroll
 * the display and code for scrolling will not be included.  This option 
 * will save some code space and one byte of RAM.
 *
 * If INCLUDE_SCROLLING is defined to be one, the scroll feature will
 * be included but not enabled.  A call to setScroll() will be required
 * to enable scrolling.
 *
 * If INCLUDE_SCROLLING is defined to be two, the scroll feature will
 * be included and enabled. A call to setScroll() will be required
 * to disable scrolling.
 */
#define INCLUDE_SCROLLING 1
```

gerardwr, 2016-01-27:
Hell Bill,

Tried the updated library and the scrolling works brilliantly.

It's possibly my "always scroll" mindset, but I had to read an re-read the (elaborate) comment on the proper use of the #define INCLUDE_SCROLLING, and was a little confused at first.

I see now that you have defaulted to "#define INCLUDE_SCROLLING 1", the behavior of the former version without scroll.

As suggested by you I changed to "#define INCLUDE_SCROLLING 2" and the displays then scrolls without a required setScroll call. Like it better this way.

I'm really happy now with your library, it's compact and easy to add to a sketch.

Can't think of another change request right now. When I do, I'll raise an issue again :-)

Thanks.

greiman, 2016-01-27:
"#define INCLUDE_SCROLLING 1"  works just like "#define INCLUDE_SCROLLING 2" if you call setScroll(true).

Otherwise it works like the old version which is the way it is used in most apps.  You are the first to want scrolling.

This example code was to illustrate use of INCLUDE_SCROLLING.
```
  #if INCLUDE_SCROLLING == 0
  #error INCLUDE_SCROLLING must be non-zero. Edit SSD1306Ascii.h
  #elif INCLUDE_SCROLLING == 1
  // Scrolling is not enable by default for INCLUDE_SCROLLING set to one.
  oled.setScroll(true);
  #else  // INCLUDE_SCROLLING
  // Scrolling is enable by default for INCLUDE_SCROLLING greater than one.
  #endif
```
If INCLUDE_SCROLLING is zero, the scrolling code is not available and some memory is saved.

If INCLUDE_SCROLLING is one, the scrolling code is available and you must call setScroll(true) to activate scrolling.

If INCLUDE_SCROLLING is two, scrolling is activated in the oled.begin() call.

closed 2016-01-27 by greiman

gerardwr, 2016-01-27:
Got it, thanks Bill!

tamsaha1995, 2017-01-10:
How can I scroll text sideways in a 128 x 64 ssd1306 display ?

greiman, 2017-01-10:
Use a different library.  This is a minimal feature library to keep it small and fast.

tamsaha1995, 2017-01-10:
Hi Grieman,
I am working on a project that would display serial data to an  OLED. I just want the serial data to print as it is on OLED (taking into consideration data will fit the screen). Each incoming buffer will get printed in a new line and it would continuously print in new line till there is availability of serial data.
I am trying to follow your "Scrollingwire" example.
How can I implement

greiman, 2017-01-10:
The only option for scrolling are set in SSD1306Ascii.h.

```
/** Set Scrolling mode for new line.
 *
 * If INCLUDE_SCROLLING is defined to be zero, new line will not scroll
 * the display and code for scrolling will not be included.  This option 
 * will save some code space and one byte of RAM.
 *
 * If INCLUDE_SCROLLING is defined to be one, the scroll feature will
 * be included but not enabled.  A call to setScroll() will be required
 * to enable scrolling.
 *
 * If INCLUDE_SCROLLING is defined to be two, the scroll feature will
 * be included and enabled. A call to setScroll() will be required
 * to disable scrolling.
 */
#define INCLUDE_SCROLLING 1
```

tamsaha1995, 2017-01-11:
Is there any support for setCursor function?

tamsaha1995, 2017-01-11:
`// Test for minimum program size.

// 0X3C+SA0 - 0x3C or 0x3D
#define I2C_ADDRESS 0x3C
#include <Wire.h>
#include "SSD1306Ascii.h"
#include "SSD1306AsciiWire.h"

SSD1306AsciiWire oled;
//------------------------------------------------------------------------------
void setup() {
  Wire.begin();                
  oled.begin(&Adafruit128x64, I2C_ADDRESS);
  oled.setFont(System5x7);
  oled.clear();
 oled.setCursor(25,25);
  oled.print("Hello world!");
}
//------------------------------------------------------------------------------
void loop() {}
`
I tried this code...but it doesn't seem to print in the 25th row.

greiman, 2017-01-11:
You need to use another library since this is a simple small library and avoids large buffers required to overcome hardware limitations of the devices.

There is no 25th row.  Read the html documentation rows are by 8-pixel units since rows on the device are addressed by bytes.

void SSD1306Ascii::setRow( uint8_t  	row	) 	
  Set the current row number.
  Parameters [in]	row	the row number in eight pixel rows.

tamsaha1995, 2017-01-11:
So why should I write to print "Hello world" at the center?

tamsaha1995, 2017-01-11:
Yeah, i figured it out.
Thanks.