greiman/SSD1306Ascii #5: "warning set400kHz() disabled for this CPU" using an ESP8266 board

gerardwr, 2016-02-08:
When using SSD1306AsciiWire.h in an Arduino sketch on an Esp8266 board the following warning is given: 
```
 #warning set400kHz() disabled for this CPU.
```
I see in SSD1306AsciiWire.h that this warning is given unless an AVR or SAM3X8E is recognized. 

Despite this warning the library works fine on my ESP8266 :-)

greiman, 2016-02-08:
I only have code to set the speed for AVR and Due.  

I put the warning in so users would not call set400kHz() and expect faster performance on other boards.

You can either remove the warning or add the correct code for ESP8266.

closed 2016-02-08 by greiman

gerardwr, 2016-02-09:
I added a setClock statement to my Arduino sketch:
```
  Wire.begin();   
  Wire.setClock(400000);
```
Ofcourse this does not eliminate the warning, but now the buildup of the Oled is considerably faster on my Wemos D1 Esp8266 board.

Thanks.