greiman/SSD1306Ascii #12: Brightness issue

nervusvagus, 2016-04-27:
Hi,
I did a few tests before writing this and I'm almost convinced that this issue relates only to this library.

I have a chinese made 128x64 Oled display that I run with software SPI.
In addition I have plenty of I2C sensors on board. 

When I run these together 7 out of 10 times I have an issue with the brightness of the OLED screen.

I tried running other libraries and their brightness seems to be ok. I set the contrast value in setup to   oled.setContrast(255); however that still doesnt effect anything.

I also had issues with the display not showing but when I connected the Ground of display to a seperate GND of Arduino that issues seems to be gone.

Is this something related to the library that someone can help with?

greiman, 2016-04-27:
The display is initialized close to the settings in this library [Adafruit_SSD1306](https://github.com/adafruit/Adafruit_SSD1306/blob/master/Adafruit_SSD1306.cppl).

I initially modified the Adafruit library but later totally rewrote this text only library keeping the Adafruit style device settings.

You might look at the setting used by [openGLCD](https://bitbucket.org/bperrybap/openglcd/wiki/Home).

eos1d3, 2016-04-27:
Hi,

setContrast works 100%. I just tried with value of 0, 127 and 255 and can confirm all work!

I used this `oled.begin(&Adafruit128x64)` to init the driver without any modification. And I have verified the data in the Adafruit128x64init with controller manual. All commands and data are correct.

SSD1306_SETPRECHARGE value also changes the brightness. Bit 7:4 is charge period and bit 3:0 is discharge period. Using 0xF1 will be the highest charge period and have highest brightness.

**WARNING**: OLED life are VERY VERY short life because of its lumen depreciation. For me, I believe never use the default value. It is too bright that it will be washed out in just 1 or 2 days of use!!!!!

To test this, display a text pattern with white on black for one day. Then invert the display color to back on white background without displaying any text. You will see the text you displayed earlier is on the screen in burned there.

By using the smaller value to contrast value, it helps a lot for this problem but the wash-out problem is never be eliminated even with value of 0. That I why I will never buy any oled TV. 

Use this oled module with very little screen turn-on time!!

boyswat, 2017-10-05:
Hi,

i setContrast to 0 or random number < 100 oled will not work, i try 127 and 255 it work but not different.

i try edit in SSD1306init.h that still doesnt effect anything.
SSD1306_SETCONTRAST, 0xff,
at Adafruit128x64init

greiman, 2017-10-05:
Does it work with the Adafruit library?

I can't really help you much since I don't have your display.

boyswat, 2017-10-05:
Hi,
i try Adafruit but it not display anything. and i try OakLED work but it still no difference i set 255 and 1. i use oled 0.96 ssd1306 i2c white.

HDBro, 2018-06-27:
My OLED screen works perfectly fine. But for the project I am working on, the screen has to be brighter then normal. Any help on that?

greiman, 2018-06-27:
It's not possible to increase the brightness much.  I have it near max.

This may help a little.
```
  oled.ssd1306WriteCmd(SSD1306_SETVCOMDETECT);
  oled.ssd1306WriteCmd(0X70);
  oled.setContrast(255);
```

VICLER, 2020-10-16:
Hi!
I'm testing my firmware on SSD1306 and SH1106 and today I noticed, that If I initialize an SH1106 display with Adafruit128x64 the display is darker. But it looks very nice! There appears something like shadow, and text looks as pseodo 3D. So my question is what should I do to get this effect by initializing the display as SH1106?
Here is SH1106 initialize as SH1106:
[image]
[image]

And here SH1106 display initialized as Adafruit128x64:
[image]
[image]