greiman/SSD1306Ascii #16: Fonts

holgerlembke, 2016-08-16:
Hi!

Nice work! Fast, small, working.

My program uses a central definition of used fonts and I was playing around with fonts, optimize for readability and other things. Some fonts only have digits, some have characters, not sure all have a full set of special chars.

So I came to the following questions:
-- is there any advised way to detect by program whether a font has a char?
-- are there more fonts avaiable or a way to create fonts?

greiman, 2016-08-16:
You can find the code for the first character and count of characters in a font like this:
```
  uint8_t first = readFontByte(System5x7 + FONT_FIRST_CHAR);
  uint8_t count = readFontByte(System5x7 + FONT_CHAR_COUNT);
  Serial.println(first);
  Serial.println(count);
```
This prints:

> 32
> 97

So the first code is 32, a blank and the font has 97 characters, codes 32 - 128.  128 is a degree symbol.

You can look for GLCD fonts and font creators.  Then make an include file in the same format as the ones in the fonts folder.  Add your font filename to allFonts.h

closed 2016-08-16 by greiman

holgerlembke, 2016-08-16:
Thanks Bill.

"GLCD font and font creator" was the missing bit of information to get me startet.