greiman/SSD1306Ascii #17: Add timeouts to i2c ?

penfold42, 2016-11-19:
Great library - thanks!

Has anyone noticed that this hangs if there's no display present ?

greiman, 2016-11-19:
Are you using Wire or the minimal AVR I2C driver?

The AVR I2C driver has two while loops that could hang.

```
  // AvrI2c.h line 120
 while (TWCR & (1 << TWSTO));

  // AvrI2c.h line 142
  while (!(TWCR & (1 << TWINT)));
```

penfold42, 2016-11-19:
Right now I'm using avri2c but I will likely have to migrate to wire on an esp8266.

for now I'll maybe add a simple counter based timeout in those tight loops.

Thanks

YiorgosTheo, 2016-11-26:
I have the same issue, thanks.
Is there some way to simply bypass the whole screen writing if a screen is not detected?

YiorgosTheo, 2016-11-26:
for now, a counter timeout works just fine....