greiman/SSD1306Ascii #19: small reorganization of files

robertoostenveld, 2017-01-03:
Thanks for this great library.

Could you consider reorganizing the code such that "src", "examples" and "library.properties" appear at the top level directory, rather than in the SSD1306Ascii/SSD1306Ascii subdirectory? 

That allows to install it like this

````cd ~/Arduino/library
git clone https://github.com/greiman/SSD1306Ascii.git
````

after which it is immediately detected as library and the examples can be selected from the example menu.

best regards,
Robert

greiman, 2017-01-03:
I plan to restructure all my libraries for the Arduino Library manager.  It will happen with the next mod of each of my libraries.

They will all be like [DigitalIO](https://github.com/greiman/DigitalIO).

greiman, 2018-06-20:
done

closed 2018-06-20 by greiman