greiman/SSD1306Ascii #22: Are umlauts supported?

T94T, 2017-07-21:
First of all, thank you very much for this great library!
Does it also support umlauts like ö,ä,ü... ? I did not find any solution.

greiman, 2017-07-21:
I did not add any extended fonts.  I intended this library to be very light weight.

I don't want to try to include features that libraries like u8glib or openGLCD has.

You could add a font if you can find a GLCD font you like.

Here is an [u8glib ](https://github.com/olikraus/u8glib/wiki/fontgroupfreeuniversal)example.

Another solution is to use a GLCD font editor and create a font with the extra characters.

I know of users that added Russian characters.

T94T, 2017-07-22:
Thank you very much for your quick reply. Is there a way to convert u8glib fonts into the GLCDFONTDECL format? I want to embed the u8g_font_chikita font. Hope you can help me.

greiman, 2017-07-22:
"GLCDFONTDECL" is just standard glcd format.  There are several tool that can import or create glcd fonts.

See [allFonts.h](https://github.com/greiman/SSD1306Ascii/blob/master/SSD1306Ascii/src/fonts/allFonts.h) for info about adding a glcd font.

I don't know of a tool that can convert u8glib to glcd.

[openGLCDFontCreator](https://bitbucket.org/bperrybap/openglcd/downloads/) can import system fonts on a pc and export them as glcd.

I created the ZevvPeep8x16 font from  a bdf file.  It was easy since Zevv Peep is fixed width so the program was simple and only worked for the 8x16 font.

T94T, 2017-07-24:
Thank you for your explanations. openGLCDFontCreator works not as expected. I'm going to take a font included by your library and apply the necessary changes for the umlauts by hand.

closed 2017-07-24 by T94T