greiman/SSD1306Ascii #23: Screen rotation

sticilface, 2017-08-03:
Is it possible to rotate the screen.  i.e. have it 180 degrees?

greiman, 2017-08-04:
No.  It would require a second set of fonts of remapping the bits in each byte of the existing font which would be slow.

PBudmark, 2017-09-09:
I would say yes, at least on my OLED 0.96" I2C-only. After normal initialization using Adafruit128x64, I send another two commands as I have my OLED rotated with the pins at bottom
`  //  Initialize dev from command table, turned on an cleared
  lcd.begin(&Adafruit128x64, I2C_ADDRESS);

  // Rotate display 180 degrees, connection pins at bottom
  lcd.ssd1306WriteCmd(SSD1306_SEGREMAP);  // colAddr 0 mapped to SEG0 (RESET)
  lcd.ssd1306WriteCmd(SSD1306_COMSCANINC);  // Scan from COM0 to COM[N –1], normal (RESET) 
`

Evantm, 2017-10-24:
In the library switch the values of COMSCANINC and COMSCANDEC. Like so:
COMSCANINC = 0xc0
COMSCANDEC = 0xc8

https://electronics.stackexchange.com/questions/273768/flip-the-image-of-oled-128x64-0-96-inch-display

greiman, 2018-06-11:
I added `void displayRemap(bool mode)` to select normal or remap mode.  

See the DisplayRotationSpi and DisplayRotationWire examples.