greiman/SSD1306Ascii #24: i2c scrolling on 32 line displays

scargill, 2017-09-09:
I'm using your WIRE example of scrolling (i2c) - and it isn't  - the numbers count, lines go down then the bottom line gets replaced for the rest of the counts - ie the screen doesn't scroll up.

The only change I made to the example was to change the 64 to 32.... as I have the smaller 32 pixel high displays.....

greiman, 2017-09-09:
Scrolling only works with 64 pixel high displays.
 
Scrolling depends on the fact that the memory in the display controller which is 64 pixels high matches the height of display.  In this case I can just set a pointer to scroll.

Scrolling on for a 32 pixel high display requires rewriting the display controller memory.  The controller memory is not readable and I don't buffer the data so it is impossible.

Sorry, the 64 pixel requirement it is not documented.

scargill, 2017-10-23:
Thanks - marvellous library - thank you.

d-a-v, 2018-08-19:
@scargill check #47

greiman, 2018-08-25:
Scrolling should now work on all displays.