greiman/SSD1306Ascii #25: Use with rtc ds3231

boyswat, 2017-10-02:
Hi. I using ur library, very good. But i have problem when use ds3231 rtc module. I use i2c oled and ds3231 use sda and scl like oled. Oled will not work if i using ds3231 at same time. Can you help me use both? Sr my english very bad. Thank you.

boyswat, 2017-10-02:
Ds3231 library i using
http://www.rinkydinkelectronics.com/library.php?id=73

DirtyEngineer, 2017-10-02:
My guess is that your oled module and rtc module are using the same I2c address. You can use this sketch (http://playground.arduino.cc/Main/I2cScanner) to scan the addresses of the oled and ds3231 to check. The ds3231 uses address 0x68 address.

greiman, 2017-10-03:
I can't help you solve I2C device conflict problems.

closed 2017-10-03 by greiman

boyswat, 2017-10-03:
I change other library ds3231 and it work. Sr for reply too late.