greiman/SSD1306Ascii #27: initializing problem?

kay0815, 2017-12-27:
Hi,
at first thank you very much for this tiny lib. I am using it with an adafruit 128x32 IIC display. It work almost perfect for me, but I have trouble to get the display running after connecting power to the arduino. I started using your lib after some examples of the ada lib. Creating a sketch with this lib is very simple and works fine until disconecting power and switch on again. Then the display shows only some waste pixels. The only possibility to get the display working again is to use a ada-lib example. After this I can use the sketch with your lib again until switching off power. Do you have any idea what is going wrong here? Thank you very much. 
kind regards, Kay

greiman, 2017-12-29:
I have no idea.  I have used an Adafruit 128x32 display.

ramast, 2019-05-02:
Maybe you are using the wrong communication?

This happened to me when I used the AvrI2C example but worked perfectly fine when I switched to the Wire128x32 example