greiman/SSD1306Ascii #30: Restructure folder layout for Add .zip library in arduino

jared0x90, 2018-01-24:
Hey there, I love this little i2c OLED library along with Arduianana's streaming 5. One thought to improve it for future uses is to move the actual project folder one level lower. As it stands I can download this repo as a zip, take out the "SSD1306Ascii" folder one level deep, then zip that up and then install it as a zip plugin library. As this project has no tagged releases otherwise, I think making this change would let people drop it into the Arduino IDE more easily.

greiman, 2018-01-24:
I plan to restructure it for the library manager soon.  First I need to test several mods for handling reset for I2C displays.

closed 2018-01-24 by greiman

jared0x90, 2018-01-25:
Ah OK, it hasn't been touched in a while so I thought it might be being left as is going forward. I forked it and made it compatible so I could reference it in my own source. Feel free to take a look or I can shoot your a PR if you have any interest.