greiman/SSD1306Ascii #32: Flickering 2.0

Cardboardboxprocessor, 2018-02-18:
Screen is flickering as if literally seeing it refresh. worsening on  right side and lower side of screen .

COde:

> #define I2C_ADDRESS 0x3C
> #include <Wire.h>
> #include "SSD1306Ascii.h"
> #include "SSD1306AsciiWire.h"
> 
> 
> SSD1306AsciiWire oled;
> 
> void setup()
> {  
>   Wire.begin();                
>   oled.begin(&Adafruit128x32, I2C_ADDRESS);
>   oled.set400kHz();  
>   oled.setFont(Iain5x7);
> }
> 
> void loop()
> {
>   oled.clear();
>   oled.println("flickering");
> }

Chip:
Arduino Nano V3.0, Elegoo Nano board CH340/ATmega328P

Screen:
MakerFocus 2pcs I2C OLED Display Module 0.91 Inch I2C SSD1306 OLED Display Module Blue I2C OLED Screen Driver DC 3.3V~5V for Arduino

greiman, 2018-02-18:
The SSD1306 has no clear command so a 128x32 display requires writing 512 bytes to display memory.

This take about 24 ms with 400 kHz I2C so your program shows extreme flickering.

closed 2018-02-18 by greiman

greiman, 2018-02-18:
I tried a SPI 128x32 display.  It does a clear in about 4 ms.

Flicker is much better.  This loop is fairly good with the 10 ms delay.

```
void loop() {
    oled.clear(); 
    oled.print("flicker");
    delay(10);
  }
```

sergevm1, 2019-12-04:
I have the same issue. Adafruit library doesn't have flickering at all. At they write to some buffer (I guess) and then show the updated screen by the display() command.
I liked this library for its very small amount of consumed memory. But with this issue I don't see a way to use it in practice.

greiman, 2019-12-05:
Clearing the entire display causes flickering.

This is why the Adafruit library has less flicker.  The Adafruit library doesn't clear areas that have not changed, it rewrites these areas with the same data.

If you selectively rewrite only areas that are modified there will be less flickering.