greiman/SSD1306Ascii #34: DISPLAYON, DISPLAYOFF

VolkerBo, 2018-03-05:
Hi,
I'm using the Library with Arduino pro mini and ESP8266 and an OLED-Display 128x64.

Everthing works so far and now I want to put the Display off over night and on at the beginning of the next day, to increase the lifetime of the OLED.

What would you recomment to do? 
Which command would be the best?

Regards
Volker

greiman, 2018-03-05:
Here is a [datasheet ](https://www.vishay.com/docs/37902/oled128o064dbpp3n00000.pdf)for a typical display.

This display has an operating life of 40,000 hours so I don't worry.

The datasheet recommends:

> (8)It's pretty common to use "Screen Saver" to extend the lifetime and Don't use fix information 
> for long time in real application. 

> (9)Don't use fixed information in OLED panel for long time, that will extend "screen burn" effect 
> time.

I guess clearing the display at night would be good.

VolkerBo, 2018-03-05:
_I guess clearing the display at night would be good._

And how would you recomment to do that?

I tried the following:
[image]

What am I doing wrong?

greiman, 2018-03-05:
Here is an example that turns the display off for four seconds then on for four seconds each time loop() is called.

It is for a 128x64 I2C display.
```

#include <Wire.h>
#include "SSD1306Ascii.h"
#include "SSD1306AsciiWire.h"

// 0X3C+SA0 - 0x3C or 0x3D
#define I2C_ADDRESS 0x3C

// Define proper RST_PIN if required.
#define RST_PIN -1

SSD1306AsciiWire oled;
//------------------------------------------------------------------------------
void setup() {
  Wire.begin();
  Wire.setClock(400000L);

#if RST_PIN >= 0
  oled.begin(&Adafruit128x64, I2C_ADDRESS, RST_PIN);
#else // RST_PIN >= 0
  oled.begin(&Adafruit128x64, I2C_ADDRESS);
#endif // RST_PIN >= 0

  oled.setFont(System5x7);
  oled.clear();
  oled.print("Hello world!");
}
//------------------------------------------------------------------------------
void loop() {
  oled.ssd1306WriteCmd(SSD1306_DISPLAYOFF);
  delay(4000);
  oled.ssd1306WriteCmd(SSD1306_DISPLAYON);
  delay(4000);    
}
```

VolkerBo, 2018-03-06:
Thank you,
these commands are better :-).
And I had to learn that they can be compiled without trouble only in LOOP and not in SETUP.