greiman/SSD1306Ascii #35: X and Y

freakout242, 2018-04-08:
I have been looking everywhere for a way to print a line to a specific X and Y but I can't seem to find the function. The closest thing I could find in the example sketches was

oled.setCursor(0, 10); 

but that doesn't seem to do anything. If anyone could help me out I would appreciate it.

dewington, 2018-04-24:
Try using oled.setCursor(0, 2);

I think row works in blocks of 8 pixels so that should set text about half way on a 32px high screen.

greiman, 2018-04-25:
The hardware addressing for SSD1306 devices is by pixel for column and 8 pixels for row.

See the documentation for member functions in Arduino/libraries/SSD1306Ascii/doc/SSD1306Ascii.html.

Here is setCursor() from the html documentation. 

> void SSD1306Ascii::setCursor(uint8_t col, uint8_trow ) 		
> 
> Set the cursor position.
> 
> Parameters
>     [in]	col	The column number in pixels.
>     [in]	row	the row number in eight pixel rows.