greiman/SSD1306Ascii #37: Working with Heltec Wifi Kit 32

ba58smith, 2018-04-24:
This library is exactly what I'm looking for, but I tried your basic example (WireI2c128x64.ino) for the Heltec Wifii Kit 32 with .96" OLED display, and I can't get it to work. It compiles and downloads to the board without a problem, but never displays anything. I'm guessing it's because there's not a DevType defined for that board, and some of the initialization settings aren't correct. Ditto for the I2C pin's hex number. My problem, being pretty new to all of this, is trying to figure out how to modify an existing DevType to work with this board, and to determine its I2C pin hex number.

This is the github page for the board itself. What do I need to know to make this board work with this library? Thanks!

Aggebitter, 2018-04-25:
// Some hints for heltec wifi lora
#include <Wire.h>
#include <U8x8lib.h>
U8X8_SSD1306_128X64_NONAME_SW_I2C u8x8(/* clock=*/ 15, /* data=*/ 4, /* reset= */ 16);

void setup{
	pinMode(16, OUTPUT);
	**digitalWrite(16, LOW);  // set GPIO16 low to reset OLED
	delay(50);
	digitalWrite(16, HIGH); // while OLED is running, must set GPIO16 in high**
	u8x8.begin();
	u8x8.setPowerSave(0);
	u8x8.setFont(u8x8_font_chroma48medium8_r);
	u8x8.setCursor(0, 0);			
	u8x8.print("Hello World!");
}

void loop {

}

ba58smith, 2018-04-25:
Aggebitter,

Thanks for that, I'll try it. But to clarify, I have the Heltec Wifi Kit 32, NOT the Heltec LoRa Kit. And, more important, I have no trouble getting the board to work with the u8x8 library. It's the SSD1306Ascii library that I can't get to work.

Aggebitter, 2018-04-25:
Have ju tryed the highlighted reset part?
If it uses the same oled noname ssd1306 as in the LoRa kit that might help. 

Had strange issues before I did a manual reset

greiman, 2018-04-25:
Looks like the Heltec Wifi Kit 32 has the OLED connected to pins SCL = 15, SDA = 4, and RST = 16.

These are not the standard Wire pins.  The Wire library uses SDA = 21 and SCL = 22.

The u8x8 library supports software I2C on any pair of pins. 

SSD1306Ascii has no support for software I2C.

ba58smith, 2018-04-26:
greiman, that would certainly explain why it wouldn't work. Would you mind implementing the software I2C sometime today? :grinning: No? OK - could I add or change a couple lines to your files somewhere to specify those different SDA and SCL pins? Or is it not that simple? Thanks so much! Your library seem so much easier to use - I'd really to be able to us it.

greiman, 2018-04-26:
> No? OK - could I add or change a couple lines to your files somewhere to specify those different SDA and SCL pins? Or is it not that simple? Thanks so much! Your library seem so much easier to use - I'd really to be able to us it.

It looks like the ESP32 Wire library allows you to specify the pins for sda and scl in the Wire.begin() call.

See this [link](https://www.esp32.com/viewtopic.php?t=2058).

Try replacing the Wire.begin() line in the HelloWorldWire example with this:

`  Wire.begin(4, 15);`

This assumes the Heltec Wifi Kit 32 uses standard ESP32 pin numbers.

If this works, it would be possible to modify SSD1306AsciiWire.h to use the second ESP32 I2C controller.  This would require reading the code for the ESP32 Wire library.

ba58smith, 2018-04-26:
Thank you so very much! I'll give that a try.

ba58smith, 2018-04-27:
greiman,

That was it! Wire.begin(4, 15) made "Hello world!" show up. Thanks so much!

SALLDHMF, 2019-07-07:
Hi.
Unfortunately, I'm not as lucky as ba58smith ;-). Replacing Wire.begin () with Wire.begin (4, 15) in HelloWorldWire did not solve the problem of displaying texts (with your library) on Heltec LoRa 32. My module is very similar to Heltec Wifii Kit 32. OLED is also connected by I2C and also via pins SDA = 4, SCL = 15, and RST = 16.

Maybe I should change something else in the HelloWorldWire code?

My / Your code:

// Test for minimum program size.

#include <Wire.h>
#include "SSD1306Ascii.h"
#include "SSD1306AsciiWire.h"

// 0X3C+SA0 - 0x3C or 0x3D
#define I2C_ADDRESS 0x3C

// Define proper RST_PIN if required.
#define RST_PIN 16

SSD1306AsciiWire oled;
//------------------------------------------------------------------------------
void setup() {
  Wire.begin(4, 15);
  Wire.setClock(400000L);

#if RST_PIN >= 0
  oled.begin(&Adafruit128x64, I2C_ADDRESS, RST_PIN);
#else // RST_PIN >= 0
  oled.begin(&Adafruit128x64, I2C_ADDRESS);
#endif // RST_PIN >= 0

  oled.setFont(System5x7);
  oled.clear();
  oled.print("Hello world!");
}
//------------------------------------------------------------------------------
void loop() {}

[WIFI_LoRa_32_V2.pdf](https://github.com/greiman/SSD1306Ascii/files/3365150/WIFI_LoRa_32_V2.pdf)

Tomasz

greiman, 2019-07-07:
I don't see an obvious problem.
 
I can't find any info about the display.  Since I don't have your hardware I can't do any tests.

SALLDHMF, 2019-07-07:
OK. I understand.
The layout of the pins was included in the PDF file in the first comment.

The display is 0.96-inch 128 * 64 dot matrix OLED display, with SSD1306 controller.

The module is https://heltec.org/project/wifi-lora-32/

What other information can help you diagnose the problem?

SALLDHMF, 2019-07-07:
https://github.com/HelTecAutomation/Heltec_ESP32

[Uploading WiFi_LoRa_32_V2(433,470-510).PDF…]()

SALLDHMF, 2019-07-07:
https://github.com/HelTecAutomation/Heltec_ESP32

[WiFi_LoRa_32_V2(433,470-510).PDF](https://github.com/greiman/SSD1306Ascii/files/3365685/WiFi_LoRa_32_V2.433.470-510.PDF)

greiman, 2019-07-08:
I looked at the schematic and other info before the last post.

I can't do more with no hardware.

This is from an example that looks for the OLED.

Add these includes:  
```
#include "Arduino.h"
#include "heltec.h"
```
replace:
```
  Wire.begin(4, 15);
  Wire.setClock(400000L);
```
With:
```
    Heltec.begin(true, false, true);
    Wire.begin(SDA_OLED, SCL_OLED); 
```

Try to run [this](https://github.com/HelTecAutomation/Heltec_ESP32/tree/master/examples/ESP32) example.

It should give the I2C address.  Maybe Heltec used the alternate SSD1306 address.

SALLDHMF, 2019-07-08:
Yes. I have "Hello world!" On the display.

Thank you so much for help.