greiman/SSD1306Ascii #38: clear() takes 60ms.

MintSoup, 2018-05-05:
oled.clear takes about 60ms to clear the display, using an I2C 128x64 from ebay. This causes horrible flickering problems. Is it possible to speed this up?

MintSoup, 2018-05-05:
Hmmm... not sure, but I think I was using 400khz. Will confirm tomorrow.

greiman, 2018-05-06:
I tried clear() on a ebay 128x64 and I get 46.3 ms.  at 400kHz.  An ebay SPI display clears in 8.1 ms.

I will experiment a bit more.  I may have tested with a processor faster than an Uno. or a 128x32 display.

Souji911, 2019-08-02:
I've got this problem as well, i have shifting lengths of text on an I2C 128x64 and it doesn't clear the leftover characters when the lines get shorter, except if i use oled.clear(); which makes the display flicker.

i have tried using oled.clear(Columnstart,Column...); but that just makes the small part selected flicker because of my rather short update delay.

Would it be possible to make an oled.clearram(); or something like that? Where it just clears the ram buffer, but doesn't update the display. Then you can fill the ram again with new stuff and then update the display.

greiman, 2019-08-03:
I repeat. There is no fast memory clear command for the SSD1306.  oled.clear() must write blanks to all locations in the SSD1306 memory so it takes a long time.

To reduce flicker, write blanks over the extra characters in each line.  Also only write portions of the display that change.

SPI displays clear much faster so the flicker is less noticeable.

See various other posts about this issue.  Others have proposed work arounds.

Souji911, 2019-08-03:
Thanks for your answer!
I actually got it working a short while after posting, using the "blanks after text" example and it works the charm!