greiman/SSD1306Ascii #43: Scrolling

tvixen, 2018-06-20:
Hi there

Would it be possible for you to add a vertical scrolling function ?

Thanks in advance.
Tvix.

greiman, 2018-06-20:
I assume you mean the 0XA3 vertical scrolling command.  Currently I have no plans to for to implement this vertical scrolling command.

Several users have implemented various types of scrolling using the `ssd1306WriteCmd()` function. 

I don't see a simple API that would cover the cases these users have implemented.

tvixen, 2018-06-22:
Btw... do you have a function to power off the display, so it will not use any power.... Or at leased a minumum of power ?

Thanks in advance.
Tvix.

greiman, 2018-06-22:
You can try the following commands to save power.

```
  oled.ssd1306WriteCmd(SSD1306_DISPLAYOFF);
  oled.ssd1306WriteCmd(SSD1306_DISPLAYON);
```

These commands switch between normal and sleep mode.  Here is the datasheet documentation.

> 10.1.12 Set Display ON/OFF (AEh/AFh)
> These single byte commands are used to turn the OLED panel display ON or OFF.
> When the display is ON, the selected circuits by Set Master Configuration command will be turned ON.
> When the display is OFF, those circuits will be turned OFF and the segment and common output are in VSS state and high impedance state, respectively. These commands set the display to one of the two states:
> o AEh : Display OFF
> o AFh : Display ON

tvixen, 2018-06-24:
Thanks.... It's working.

d-a-v, 2018-08-19:
@tvixen check #47