greiman/SSD1306Ascii #44: Characters are displayed 1 pixel off screen to the left

HuubBuis, 2018-06-20:
This is the first library that looks promising. But on the left, 2 columns of pixels are missing and on the right of the display there is a 2 pixel white column. 

Sketch: WireI2c128x64
Oled: i2C 128 x 64 white from eBay

greiman, 2018-06-20:
Most likely it is not a SSD1306 controller. It might be a SH1106 controller.

Try  replacing `&Adafruit128x64` by `&SH1106_128x64` in the `begin()` call like this.

```
#if RST_PIN >= 0
  oled.begin(&SH1106_128x64, I2C_ADDRESS, RST_PIN);
#else // RST_PIN >= 0
  oled.begin(&SH1106_128x64, I2C_ADDRESS);
#endif // RST_PIN >= 0
```

HuubBuis, 2018-06-20:
Wow,

 

That did it,

 

Thanks for the support

 

Best regards,

 

Huub Buis

 

  _____  

screen to the left (#44)

 

it on GitHub, or mute
ZoU6aI3ks5t-o6SgaJpZM4Uvt3e>  the thread.
<https://github.com/notifications/beacon/AV4i2Hcp1Gj0tE_VFH3ZVyUIh6a8lvyMks5
t-o6SgaJpZM4Uvt3e.gif>

asokolsky, 2018-06-26:
Thank you!  Solved my problem as well.
Agreed, this is the most promising library for use with OLEDs from eBay.  What attracts me most are fonts and simple API.
Thanks!

classic-audio, 2018-08-06:
Thank you! .......... and solved my problem too.

John-Lluch, 2018-12-24:
You really need to write a Wiki for your excellent library.   I didn't realise your library was compatible with the SH1106 displays until I looked deeply into it. I was using U8x8lib before but yours is infinitely better!. If only you took some time to fully document it...  :-)

Flipperkastje, 2020-10-18:
Thank you Bill Greiman. This solved it for me !

IvyBridge1, 2022-02-22:
Thanks Bill Greiman.  This worked and solved the problem.  ;-))