greiman/SSD1306Ascii #50: Added Support for 96x16 SSD1306 Oled Display

DirtyEngineer, 2018-09-22:
I added support for the 96x16 SSD1306 Oled Display. Initialization follows specification from this displays [data sheet](https://www.buydisplay.com/download/manual/ER-OLED0.69-1_Series_Datasheet.pdf).

**Adafruit96x16** 
[SSD1306init.zip](https://github.com/greiman/SSD1306Ascii/files/2407565/SSD1306init.zip)

It's not actually an Adafruit display but it seem like it would be better to conform to how the rest of SSD1306 displays are initialized in this library.

> //------------------------------------------------------------------------------
// This section is based on data sheet for the 96x16 SSD1306  
// https://www.buydisplay.com/download/manual/ER-OLED0.69-1_Series_Datasheet.pdf
/** Initialization commands for a 96x16 SSD1306 oled display. */
static const uint8_t MEM_TYPE Adafruit96x16init[] = {
    // Init sequence for Adafruit 128x32 OLED module
    SSD1306_DISPLAYOFF,
    SSD1306_SETDISPLAYCLOCKDIV, 0x80,  // the suggested ratio 0x80
    SSD1306_SETMULTIPLEX, 0x0F,        // ratio 32
    SSD1306_SETDISPLAYOFFSET, 0x0,     // no offset
    SSD1306_SETSTARTLINE | 0x40,       // line #0
    SSD1306_CHARGEPUMP, 0x14,          // internal vcc
    SSD1306_MEMORYMODE, 0x00,          // page mode
    SSD1306_SEGREMAP | 0xA1,           // column 95 mapped to SEG0
    SSD1306_COMSCANDEC,                // column scan direction reversed
    SSD1306_SETCOMPINS, 0x02,          // sequential COM pins, disable remap
    SSD1306_SETCONTRAST, 0xAF,         // contrast level 175
    SSD1306_SETPRECHARGE, 0xF1,        // pre-charge period (1, 15)
    SSD1306_SETVCOMDETECT, 0x20,       // vcomh regulator level
    SSD1306_DISPLAYALLON_RESUME,
    SSD1306_NORMALDISPLAY,
    SSD1306_DISPLAYON
};
/** Initialize a 96x16 SSD1306 oled display. */
static const DevType MEM_TYPE Adafruit96x16 = {
  Adafruit96x16init,
  sizeof(Adafruit96x16init),
  96,
  16,
  0
};

[image]

greiman, 2018-09-29:
Added SSD1306_96x16 display.

It is based on code from https://github.com/olikraus/u8g2/blob/master/csrc/u8x8_d_ssd1306_96x16.c

It is really identical to your code but I found it earlier.

DirtyEngineer, 2018-10-01:
Thanks and Cheers!

closed 2018-10-01 by DirtyEngineer