greiman/SSD1306Ascii #51: Extremly slow refresh time

machmar, 2018-10-30:
The display is really flickering. If you want my code, let me know. Thanks for response

krukhlis, 2018-10-30:
Try to use i2c driver from greiman instead of slow buggy Wire.
I mean run this example https://github.com/greiman/SSD1306Ascii/blob/master/examples/AvrI2c128x64/AvrI2c128x64.ino.
Also, call the method to increase i2c frequency to 400 000 instead of default slow one.

DirtyEngineer, 2018-10-31:
My guess is that you have a clear display call in your main loop. This will cause flickering.

machmar, 2018-10-31:
I forgot to mention that I am using wire but all the demos work well. Not my code though. I was thinking if there is any way to write multiple lines at once. code here: `
#include <EEPROM.h>
#include <Rotary.h>
#include <OneButton.h>
#include <Wire.h>
#include <Adafruit_SSD1306_STM32.h>
#include <Adafruit_GFX.h>
#include <Fonts/FreeSans12pt7b.h >
#include <Fonts/FreeSansBold24pt7b.h >

Adafruit_SSD1306 display(-1);
Rotary rotary = Rotary(PA2, PA3);
OneButton coderButton(PA8, true);

int menuEncoderVar1 = 0;
int menuEncoderVar2 = 0;
int menuEncoderVar3 = 0;
int menuEncoderVar4 = 0;
int menuEncoderVar5 = 0;
int menuEncoderVar6 = 0;
int menuEncoderVar7 = 0;
int menuEncoderVar8 = 0;
int menuVar1 = 0;
int normalClick = 0;
int menuScrollVar1 = 0;

int brightness = 25;
int brightnessVar = 0;
int animationSpeed = 10;
int animationSpeedVar = 0;
int numberPixels = 0;
int animationColorWheel = 0;

long currentmicros = 0;
long previousmicros = 0;
long currentmicros1 = 0;
long previousmicros1 = 0;
int topClock = 0;
int topClockStop = 0;
int floDirection = 0;
int floNumber = 0;

void rotate() {
  unsigned char result = rotary.process();
  if (menuVar1 == 0) {
    if (result == DIR_CW) menuEncoderVar1++;
    else if (result == DIR_CCW)menuEncoderVar1--;
  }
  if (menuVar1 == 2) {
    if (result == DIR_CW) menuEncoderVar2++;
    else if (result == DIR_CCW)menuEncoderVar2--;
  }
  if (menuVar1 == 4) {
    if (result == DIR_CW) menuEncoderVar3++;
    else if (result == DIR_CCW)menuEncoderVar3--;
  }
  if (menuVar1 == 6) {
    if (result == DIR_CW) menuEncoderVar4++;
    else if (result == DIR_CCW)menuEncoderVar4--;
  }
  if (menuVar1 == 8) {
    if (result == DIR_CW) menuEncoderVar5++;
    else if (result == DIR_CCW)menuEncoderVar5--;
  }
  if (menuVar1 == 10) {
    if (result == DIR_CW) brightness++;
    else if (result == DIR_CCW)brightness--;
  }
  if (menuVar1 == 12) {
    if (result == DIR_CW) brightness++;
[... 497 lines cut ...]
    if (floNumber >= 5) floDirection = !floDirection;
    if (floNumber <= 0) floDirection = !floDirection;
    display.setCursor(floNumber, menuEncoderVar6 * 10 + 18);
    display.println(">");

    if (menuEncoderVar6 == 0 && normalClick == 1) {
      animationColorWheel = 1;
      menuVar1 = 6;
      normalClick = 0;
    }
    if (menuEncoderVar6 == 1 && normalClick == 1) {
      menuVar1 = 100;
      animationColorWheel = 0;
      normalClick = 0;
    }
    if (menuEncoderVar6 == 2 && normalClick == 1) {
      menuVar1 = 6;
      normalClick = 0;
    }
    if (menuEncoderVar6 >= 3) menuEncoderVar6 = 2;
    if (menuEncoderVar6 <= -1) menuEncoderVar6 = 0;
  }

display.display();
}`

DirtyEngineer, 2018-10-31:
```
void loop() {

..
..
display.clearDisplay();
..
..
```

You're clearing the display each time through your loop.

greiman, 2018-10-31:
Closed - not this library.
```
#include <Adafruit_SSD1306_STM32.h>
#include <Adafruit_GFX.h>
```

closed 2018-10-31 by greiman

machmar, 2018-11-03:
sorry I sent the wrong code, please reopen this issue

machmar, 2018-11-03:
`#include <SSD1306Ascii.h>
#include <SSD1306AsciiWire.h>
#include <EEPROM.h>
#include <Rotary.h>
#include <OneButton.h>
#include <Wire.h>
SSD1306AsciiWire oled;
Rotary rotary = Rotary(PA2, PA3);
OneButton coderButton(PA8, true);

int menuEncoderVar1 = 0;
int menuEncoderVar2 = 0;
int menuEncoderVar3 = 0;
int menuEncoderVar4 = 0;
int menuEncoderVar5 = 0;
int menuEncoderVar6 = 0;
int menuEncoderVar7 = 0;
int menuEncoderVar8 = 0;
int menuVar1 = 0;
int normalClick = 0;
int menuScrollVar1 = 0;

int brightness = 25;
int brightnessVar = 0;
int animationSpeed = 10;
int animationSpeedVar = 0;
int numberPixels = 0;
int animationColorWheel = 0;

long currentmillis = 0;
long previousmillis = 0;
long currentmillis1 = 0;
long previousmillis1 = 0;
int topClock = 0;
int topClockStop = 0;
int floDirection = 0;
int floNumber = 0;

void rotate() {
  unsigned char result = rotary.process();
  if (menuVar1 == 0) {
    if (result == DIR_CW) menuEncoderVar1++;
    else if (result == DIR_CCW)menuEncoderVar1--;
  }
  if (menuVar1 == 2) {
    if (result == DIR_CW) menuEncoderVar2++;
    else if (result == DIR_CCW)menuEncoderVar2--;
  }
  if (menuVar1 == 4) {
    if (result == DIR_CW) menuEncoderVar3++;
    else if (result == DIR_CCW)menuEncoderVar3--;
  }
  if (menuVar1 == 6) {
    if (result == DIR_CW) menuEncoderVar4++;
    else if (result == DIR_CCW)menuEncoderVar4--;
  }
  if (menuVar1 == 8) {
    if (result == DIR_CW) menuEncoderVar5++;
    else if (result == DIR_CCW)menuEncoderVar5--;
  }
  if (menuVar1 == 10) {
    if (result == DIR_CW) brightness++;
    else if (result == DIR_CCW)brightness--;
  }
  if (menuVar1 == 12) {
    if (result == DIR_CW) brightness++;
    else if (result == DIR_CCW)brightness--;
  }
  if (menuVar1 == 31) {
    if (result == DIR_CW) menuEncoderVar6++;
[... 440 lines cut ...]
      if (floDirection == 1) floNumber--;
    }
    if (floNumber >= 5) floDirection = !floDirection;
    if (floNumber <= 0) floDirection = !floDirection;
    oled.setCursor(floNumber, menuEncoderVar6 + 2);
    oled.println(">");

    if (menuEncoderVar6 == 0 && normalClick == 1) {
      animationColorWheel = 1;
      menuVar1 = 6;
      normalClick = 0;
    }
    if (menuEncoderVar6 == 1 && normalClick == 1) {
      menuVar1 = 100;
      animationColorWheel = 0;
      normalClick = 0;
    }
    if (menuEncoderVar6 == 2 && normalClick == 1) {
      menuVar1 = 6;
      normalClick = 0;
    }
    if (menuEncoderVar6 >= 3) menuEncoderVar6 = 2;
    if (menuEncoderVar6 <= -1) menuEncoderVar6 = 0;
  }
}`

greiman, 2018-11-03:
Looks like you clear the screen each time you enter loop so it will flicker.

Clearing the screen requires writing 1024 bytes of zero to the display memory.

oled,clear() takes about 46 ms. 

Run this code in loop.  It flickers.

```
void loop() {
  uint32_t m = micros();
  oled.clear();
  oled.print("\nmicros: ");
  oled.print(micros() - m);
  }
```

noeldum, 2024-09-02:
I am facing the same issue as the OP here.

With the Adafruit lib I do clear the display in a loop and it does not flicker.

Using the same code ported to SSD1306Ascii it does flicker. 

What is the reason for this? Is it because one is using buffer and the other not?

What are the options to avoid this? Should I use clearField() to just clear the area of interest. The problem with that is that I need to know the length of what I am displaying.

greiman, 2024-09-02:
> What is the reason for this? Is it because one is using buffer and the other not?

This is likely the reason. SSD1306Ascii does not use a buffer image of the screen.

It will be much faster if you only clear areas that you are not rewriting.

You can use this function to find the length:

` size_t strWidth(const char* str) const;`