greiman/SSD1306Ascii #58: Adafruit5x7.h has a missing character '▓' when using full the character set

hlide, 2019-03-23:
▓ is missing in the line 197. It makes the following characters with a wrong ascii number, so the last character in the font is 253 instead of 254.

hlide, 2019-03-23:
And as a side-note, the Arduino Libraries seems to have an old version with no invert mode.

greiman, 2019-03-23:
> ▓ is missing in the line 197. It makes the following characters with a wrong ascii number, so the last character in the font is 253 instead of 254.

Looks like the latest Adafruit version has the missing character plus other changes.  Guess I should use the new version unless there is a reason not to.

Here is a diff:
```
151c151
< 	0x21, 0x54, 0x54, 0x78, 0x41, 
---
> 	0x22, 0x54, 0x54, 0x78, 0x42, // a-umlaut
161c161
< 	0xF0, 0x29, 0x24, 0x29, 0xF0, 
---
> 	0x7D, 0x12, 0x11, 0x12, 0x7D, // A-umlaut
167c167
< 	0x32, 0x48, 0x48, 0x48, 0x32, 
---
> 	0x3A, 0x44, 0x44, 0x44, 0x3A, // o-umlaut
172c172
< 	0x39, 0x44, 0x44, 0x44, 0x39, 
---
> 	0x3D, 0x42, 0x42, 0x42, 0x3D, // O-umlaut
195,196c195,197
< 	0xAA, 0x00, 0x55, 0x00, 0xAA, 
< 	0xAA, 0x55, 0xAA, 0x55, 0xAA, 
---
> 	0x55, 0x00, 0x55, 0x00, 0x55, // #176 (25% block) missing in old code
> 	0xAA, 0x55, 0xAA, 0x55, 0xAA, // 50% block
> 	0xFF, 0x55, 0xFF, 0x55, 0xFF, // 75% block
243c244
< 	0x7C, 0x2A, 0x2A, 0x3E, 0x14, 
---
> 	0xFC, 0x4A, 0x4A, 0x4A, 0x34, // sharp-s or beta
```
 
> And as a side-note, the Arduino Libraries link seems to have an old version with no invert mode.

The Arduino library manager seems to have the latest version, 1.2.3 which has invert.

greiman, 2019-03-24:
See 1.2.4. I copied latest extended characters from Adafruit GFX.

hlide, 2019-03-24:
Thanks. Ultimately, I will create my own fonts for specific symbols unless fonts cannot be switched dynamically without hurting performance. I wanted to use that font temporarily for a progress bar using progressive filling blocks.

greiman, 2019-03-24:
I do a progress bar with raw writes to the display memory like this:
```
  for (int i = 0; i < 127; i++) {
    oled.ssd1306WriteRam(7);
    delay(200);
  }
```
It's three pixels high and advances one pixel per write.

hlide, 2019-03-24:
Thanks for the tip! it would definitely be better to use it that way and probably for special symbols as well where a font would be overkill indeed.

closed 2019-04-06 by hlide