greiman/SSD1306Ascii #60: println with "clear to end of row"

Humancell, 2019-05-02:
I'm looking at the best way to create an enhanced "println" that would also clear to the end of the row.  I'm using the different rows to display different information, and update it every second.  If the length of the string is longer, then shorter, it leaves debris at the end of the row from the prior string.

In looking at the code it seems that I could use clearToEOL, but then I need to know where the last pixels were used by the println ... and I can't quite figure that out.

Any suggestions on how to accomplish this?  I'm obviously trying to avoid clearing the row before writing to it.

d-a-v, 2019-05-03:
The call `::clearToEOL()` is in the library and also automatically called when you `::println()`.

greiman, 2019-05-03:
There are several ways to handle this problem.  Since you didn't post code, I can't comment on your enhanced println().

I suspect you could call ::clearToEOL() before calling println().  This will just clear the rest of the line.

The SixAdcFieldsWire and SixAdcFieldsSpi examples use ::clearField() to position the cursor to a field and clear it.

The ProportionalFormWire and ProportionalFormSPI handles forms with proportional fonts.

krukhlis, 2019-05-03:
Joining party. :)
> I suspect you could call ::clearToEOL() before calling println(). This will just clear the rest of the line and position to the begining of the next line.
BTW, if I do this way -- screen flickers. 
I output 6 lines to OLED 128x64 using 
clearToEOL() followed by print() -- unfortunately, screen flickers every draw call. :(

greiman, 2019-05-03:
Did you  print the text , call clearToEol, then println() like this?

```
  oled.print("The text for this line");
  oled.clearToEol();  // clear rest of line
  oled.println();  // go to next line
```
This should minimize flicker.

You could remember the end column for each line and print spaces something like this:
```

  uint8_t previousCol[6] = {0};
 
 // do this for each line where i is the line index.

  oled.print("new text for line i");
  uint8_t saveCol = oled.col();
  while (oled.col() < previousCol[i])  {
    oled.print(' ');
  }
  oled.println();  
  previousCol[i] = saveCol;
```

krukhlis, 2019-05-06:
Hi guys,
> oled.print("The text for this line");
>   oled.clearToEol();  // clear rest of line
>   oled.println();  // go to next line

I am using slightly different approach:
a) I am using SSD1306AsciiAvrI2c driver
b) my "menu" items can share the same row, just different horizontal position
c) Items are sorted by row
d):
...
int last_row = infoItem[0].row; //this one always exists and has a title of menu
for (int i =0;i<INFO_SIZE;i++){
   if( last_row !=infoItem[i].row) {
       oled.clearToEol();  // clear to the end of previous row, we have finished previous row
     }
   oled.setCursor(infoItem[i].col, infoItem[i].row);
   oled.print(infoItem[i].infoItem);
   last_row = infoItem[i].row;
}

where infoItem is an array of Structs that define my "menu".
Unfortunately, this code flickers. When I comment out oled.clearToEol() call -- no flickering at all. :(

greiman, 2019-05-06:
Sounds like clearToEol() is not the best answer if you are just printing a number in each field.

Just print blanks to fill the field.  oled.print(infoItem[i].info) returns the number of characters printed so use that number to calculate how many blanks to print.
```
  nClear = fieldWidth - oled.print(infoItem[i].info);
  // print nClear blanks here
```
You could also use oled.clearField(oled.col(), oled.row(), nClear).

Humancell, 2019-05-06:
Hello,

Thank you for all of the responses ...

> The call `::clearToEOL()` is in the library and also automatically called when you `::println()`.

Is this statement actually true?  If so, it does not seem to be working for me.

I'll give some of these other suggestions a try ... and post my results.

greiman, 2019-05-06:
clearToEOL is not called by println().

println() is in the Arduino Print class  so it does no call clearToEOL().

The only call to clearToEOL() is when scrolling is enable and text is scrolled.  println() will cause scrolling in this case.

d-a-v, 2019-05-06:
>  println() can cause scrolling in the case when the cursor is on the last line.

Well, sorry for this imprecision :]

krukhlis, 2019-05-07:
Ok, guys, results of my tests:
1) clearToEOL() -- flickers
2) clearField() --flickers a little bit
3) padding item with spaces on item's right -- no flickering at all!
For me -- the 3rd solution is best.
I hope it can help some people.
Thanks.