greiman/SSD1306Ascii #61: Using wire.h runs the I2C at 550k

JonRobert, 2019-05-17:
First I must say this is an awesome driver :) and I'm thankful Bill has chosen to share it.

Summary:  Using "wire.h" and oled.begin with &SH1106 the I2C bus runs at approx 550KHz.

I started from the AvrI2c128x64.ino and modified it of my needs.  I started with the original "SSD1306AsciiAvrI2c.h" then changed to the "wire.h" to be compatible with my BME680 sensor.
I'm using a 16Mhz Pro mini.
I downloaded the master zip file on 2019-05-12
I'm using the "SH1106_128x64"

The issue I had was when I switched to the wire.h method, the I2C bus was running at ~550Khz!
In the below code you can see I tried a number of commands to set the frequency, Finally the addition of "TWBR = 72;" worked.

I read the #48 post ".... slow I2C" but could not not get any of the suggestions to work.
In SSD1306Ascii.h I changed:
#define OPTIMIZE_I2C 0
#define AVRI2C_FASTMODE 0

While I have a solution (it seems) I find it less than elegant, or I don't know what I'm doing (the most likely).

Is there a better solution?

`
#include <Wire.h>
#include "SSD1306Ascii.h"
#include "SSD1306AsciiWire.h"

#define I2C_ADDRESS 0x3C	// 0x3C alt add = 0x3D
SSD1306AsciiWire oled;		// create an "oled" object

//------------------------------------------------------------------------------
// Setup displays variable names and units.

void setup() {

  //Wire.twi_setFrequency(100000);	//'class TwoWire' has no member named 'twi_setFrequency'
  //oled.setI2cClock(100000);		//'class SSD1306AsciiWire' has no member named 'setI2cClock'
  TWBR = 72;						// works, I2C set to 100k
  oled.begin(&SH1106_128x64, I2C_ADDRESS);
  oled.setFont(ZevvPeep8x16);  // results in 4 lines X 16 characters
  oled.clear();

// Fill display so we can see what is being cleared.
  oled.println("----------------");
  oled.println("----------------");
  oled.println("----------------");
  oled.println("----------------");
  delay(1000);
`

greiman, 2019-05-17:
Add Wire.begin() like this.  I do not call Wire.begin() in the versions of SSD1306Ascii that use Wire since I don't know which other I2C devices are used.   I am surprised it works at all without the Wire.begin() call.

```
  Wire.begin();
  oled.begin(&SH1106_128x64, I2C_ADDRESS);
```
You can add a Wire.setClock() call after the Wire.begin() call if you want a rate other than 100kHz.

JonRobert, 2019-05-17:
Bingo!  Works great now.
Thank you for both your library and phenomenal support :) :) :)

I was surprised the display worked at all at the 550KHz rate.  Had noticed the clear function did not work well, especially when I moved the wires (currently on a wireless breadboard).  This is one of the 4 wire 1.3" oled displays from eBay.

closed 2019-05-17 by JonRobert