greiman/SSD1306Ascii #64: Max length of a column/line?

picontrolit, 2019-09-10:
Is it possible to automatically CR/LF if a certain line length is reached?
Cheers, Alex

greiman, 2019-09-10:
No.

I don't think doing an auto CR/LF is very useful since it would split words/numbers.

I can't solve the word wrap problem since print gives me no look ahead.  I get one character at a time with write calls.

picontrolit, 2019-09-11:
For me it would not matter if words or numbers are "split". I am simulating and old school 8bit homecomputer. A simple program like "10 PRINT "ALEX ";:GOTO 10 at the moment writes to a single line. It should rather behave like on a C64 when it reaches col 40 jumping to the next line (col 1).
Any idea how to achieve that with your lib would be welcome.

greiman, 2019-09-11:
The best place to insert a CR/LF would be in the calling routine in your simulator.

write(char c) processes one character at a time so it is not easy to insert a CR/LF at this level.

greiman, 2019-09-11:
Here is an example with an auto CR/LF class.  It is not well debugged and you may want to modify the functionality.

```
#include <Wire.h>
#include "SSD1306Ascii.h"
#include "SSD1306AsciiWire.h"

// 0X3C+SA0 - 0x3C or 0x3D
#define I2C_ADDRESS 0x3C

// Define proper RST_PIN if required.
#define RST_PIN -1

SSD1306AsciiWire oled;
//----------------------------------------------------------------------
// auto CR/LF class
class AutoCRLF : public Print {
  public:
  size_t write(uint8_t c) {
    size_t rtn = 0;
    if (c == '\n' || c == '\r') {
      col = 0;
      return oled.write(c);
    }  
    if (col > 20) {
      // acts as CR/LF
      oled.write('\n');
      col = 0;
      rtn = 1;
    }
    col++;
    return rtn + oled.write(c);
  }
  uint8_t col = 0;
};
AutoCRLF oledAuto;
//------------------------------------------------------------------------------
void setup() {
  Wire.begin();
  Wire.setClock(400000L);

#if RST_PIN >= 0
  oled.begin(&Adafruit128x64, I2C_ADDRESS, RST_PIN);
#else // RST_PIN >= 0
  oled.begin(&Adafruit128x64, I2C_ADDRESS);
#endif // RST_PIN >= 0
  oled.setFont(Adafruit5x7);
  oled.clear();
  oled.setScrollMode(SCROLL_MODE_AUTO);
}

void loop() {
  oledAuto.print("ABC");
  delay(200);
} 
```

b1n0-kun, 2023-12-18:
Dear greiman.
I've just tested your oledAuto class (a very simple long line) and it worked like a charm.
Can you show me how to put that oledAuto snippet into library file (not so prefer to define same class in every project tho)
Thank you !

greiman, 2023-12-18:
Look at [this](https://docs.arduino.cc/learn/contributions/arduino-creating-library-guide) tutorial.