greiman/SSD1306Ascii #67: [Q] Centered text

leikoilja, 2020-04-10:
Hey, @greiman! 

Thank you for amazing library! 

I am using a simple SSD1306 screen with you library and was wondering if there is a method/example how to print text centered?
For example something like `oled.print("Hello world", align="center"`)` maybe not specifically like that, but something similar? 

Thanks in advance

greiman, 2020-04-10:
You can center a text string by using this function:
`size_t SSD1306Ascii::strWidth(const char* str)`
It returns the width of a string in pixels.

Subtract the string width from the screen width and use setCursor so half of the pixels are before the string.

mtak, 2021-02-10:
As a quick usage example:

```
  size_t size = oled.strWidth(instrumentNameBuffer);
  oled.setCursor((oled.displayWidth()-size)/2, 0);
  oled.println(instrumentNameBuffer);
```

yusufpraditya, 2022-12-21:
how to center a text vertically?