greiman/SSD1306Ascii #68: Various examples don't work

johncblacker, 2020-04-12:
I've tied the Example scrolling display for 64 pixel high display and I get compile errors Class SSD1306AsciiWire has no member named setScrollMmode, I've tried other examples and get other compile errors.  I'm simply copying your example code from Github over to my arduino ide and trying to run.

greiman, 2020-04-12:
Yes there is no member named setScrollMmode.  Try setScrollMode.

The example [ScrollingWire](https://github.com/greiman/SSD1306Ascii/tree/master/examples/ScrollingWire) compiles OK for me.  It haas this line:
```
  // Set auto scrolling at end of window.
  oled.setScrollMode(SCROLL_MODE_AUTO);
```

johncblacker, 2020-04-12:
I mis-typed when I created the post.  Like I said, I did a copy and paste directly from your example and it didn't work - compile error!

DirtyEngineer, 2020-04-12:
The examples compile without issue for me.

@johncblacker 

Please enable the "Show verbose output during: Compilation" in File/Preferences and post your full error.