greiman/SSD1306Ascii #70: feature request: support UTF characters using print()

VICLER, 2020-06-05:
This library is awesome! 
but I have a issue printing UTF characters on the screen. I added a new font (created with FontCreator) only with cyrillic characters (starting from 1040, used chars 64).
I also added this font to allFonts.h and I have no compile errors with this font. But if I use oled.print("Привет мир!") (russian hello world) the screen stays black..
I think this font is correct, I compared it with standard fonts in library. The problem is obviously that print() do not recognizes chars at range 1040 - 1104. 
U2glib has extra function for enabling UTF support for Arduino print() function. 
Can you please add this ability for supporting UTF?
It would be the best lightweight library for whole world and not only for US and EU :D

greiman, 2020-06-05:
Use another library.  

I called this library SSD1306Ascii  since is is a very compact library that only supports ASCII.

closed 2020-06-05 by greiman

VICLER, 2020-06-05:
Your library can display UTF, right? It can also read single chars (0-255). Ascii using 32-126. Do you think it will be possible to place the cyrillic symbols to 127 - 255 and create some function that maps the print() input char to this utf space?

greiman, 2020-06-05:
You could make a font that uses additional codes.  I will not add it to fonts and I won't fix problems.

The Adafruit5x7 font supports extra codes if you edit it and change the define to:

// Restrict to 96 characters
#define ADAFRUIT_ASCII96 0

Modify the HelloWorldWire example like this to see some of the higher codes.

  oled.setFont(Adafruit5x7);
  oled.clear();
  oled.print("Hello world!");
  for (uint8_t r = 1; r < 6; r++) {
    oled.setCursor(0, r);;
    for (uint8_t c = 0; c < 25; c++) {
      oled.write(128 + 25*r +c);
    }
  }

You can make your own library of fonts.

Make a library with your font files like this.

I called the library MyFonts and it has two file.

ls -l MyFonts
total 9
-rwxrwx---+ 1 bill None   84 May 30 10:46 MyFonts.h
-rwxrwx---+ 1 bill None 4693 May 30 10:47 MySystem5x7.h

MyFonts.h just includes all your modified fonts.

In my example I just renamed System5x7 to be MySystem5x7 like this:

GLCDFONTDECL(MySystem5x7) = {

Here is MyFonts.h

#ifndef MyFonts_h
#define MyFonts_h
#include "MySystem5x7.h"
#endif  // MyFonts_h

In your program include your library like this:

#include <Wire.h>
#include "SSD1306Ascii.h"
#include "SSD1306AsciiWire.h"
#include "MyFonts.h"     //  <<-- your font library.

Just use the name for your font like this:

  oled.setFont(MySystem5x7);