greiman/SSD1306Ascii #72: SSD1306Ascii is awesome! 👍

urish, 2020-08-10:
Hi Bill,

First of all, thank you very much for creating this library! It makes it so much easier when you just want to print some text on your SSD1306 display :-)

And I really liked the fact that you have so many fonts to choose from, and also played around with the ticker functionality, it is really fun!

I added the library to an Arduino simulation engine I'm working on, so now people can use it in the [SSD1306 playground](http://localhost:4200/playground/ssd1306), as you can see below:

[image]

It shows the scrolling display example (with just a small change, I2C address changed 0x3d). 

If you are interested, I can create simulation pages for the SSD1306Ascii examples and send a pull request to add links to the README.md file, so your users can easily find them.

This means that users will be able to experiment with the library right within the browser, without having to download or install anything, and also verify their code before uploading it to the actual hardware.

What are your thoughts?

greiman, 2020-08-10:
I am not adding any features to my libraries.  I will soon stop maintaining them.

closed 2020-08-10 by greiman

urish, 2020-08-10:
Thanks for the quick response and being upfront with you plans. Does it also apply for the SdFat library?

Are you looking for a new maintainer?

greiman, 2020-08-10:
I will be fixing bugs for some time but not adding features.

urish, 2020-08-10:
Got it! If you need any help with testing or anything, feel free to drop me a line.

And if at any point you'll be interested, my suggestion to create a simulation environment where people can experiment with SdFat / SSD1306Ascii and run the code examples in the browser is still open...

DirtyEngineer, 2020-08-12:
@greiman Thank you for all your work on SSD1306Ascii!