greiman/SSD1306Ascii #74: lcdnums fonts are missing "space" in character set

gitterman, 2021-01-06:
Trying to print right adjusted numbers with the lcdnums fonts is quite difficult as there is no "space" i.e. "blank" character in the set.

greiman, 2021-01-07:
Space is not missing.  Fonts are required to be a contiguous set of characters  by the glcd font standard. Here is a comment from the creator of the font.

> This font is not a complete character set. The font only contains the characters: '+', ',', '-', '.', '/', '0', '1', '2', '3', '4', '5', '6', '7', '9', ':'

gitterman, 2021-01-07:
I understand the issue and for this I have modified the lcdnums14x24 font to contain 10 dummy characters so that I can reach the space at 0x20. 
This makes the font slightly larger but allows simpler coding.
I use your great library to show the weight on a DIY scale and I convert the float from the HX711 module with **dtostrf( weight, 7, 1, strBuf )** so that I can use overwriting at a fixed position which gives me the least flicker on the display.
Without the 'space' character I would use much more code to get the display digits right justified.

closed 2021-01-09 by gitterman