greiman/SSD1306Ascii #76: Documentation is not rendering nicely

Miq1, 2021-02-01:
Unfortunately, the HTML docs are not rendered nicely - all I could achieve is opening single HTML files to read them one by one. Am I missing something?

Plus, a picture of the look of all fonts would be a nice addition.

Miq1, 2021-02-01:
As a remedy, I found [this](https://stackoverflow.com/questions/6551446/can-i-run-html-files-directly-from-github-instead-of-just-viewing-their-source) tip on Stackoverflow. With it, the docs can be seen as

[https://raw.githack.com/greiman/SSD1306Ascii/master/doc/html/class_s_s_d1306_ascii.html](https://raw.githack.com/greiman/SSD1306Ascii/master/doc/html/class_s_s_d1306_ascii.html)

greiman, 2021-02-01:
You can got to the docs folder and click on this file.

Arduino/libraries/SSD1306Ascii/doc/SSD1306Ascii.htm

Miq1, 2021-02-01:
No, that will only give me
[image]

greiman, 2021-02-01:
What browser are you using?  If I open SSD1306Ascii.html with chrome, firefox, edge, or explorer, I get the correct result.

Miq1, 2021-02-01:
Firefox 85.0 on Windows 10. The local copy is what I tried first, but then the page is lacking all decoration and pictures:
[image]

Miq1, 2021-02-01:
Argh. My bad... 😊 

I opened the file within the ZIP - after unpacking the structure into a proper directory tree it works out fine. Sorry for having bothered you!

Miq1, 2021-02-01:
👍 
Thanks a lot!