greiman/SSD1306Ascii #78: Atmega328PB SSD1306AsciiAvrI2c issues

ZL1CVD, 2021-04-27:
FYI - some end user feedback for you.

I have had huge stability problems when using SSD1306AsciiAvrI2c.h

Display starts then randomly everything locks up. Tried a wide range of speeds using oled.setI2cClock which did prove some speeds withing a range worked better than others.

Changed code to use Wire SSD1306AsciiWire.h and all issues resolved EXCEPT now have low memory. But it does work.

greiman, 2021-04-27:
Wire has lots of timeouts and resets so it tends not to hang with marginal devices.  There are two places in AvrI2c that could hang and I plan to add a timeout and reset to these.   Unfortunately I don't have a marginal device to verify the fix.

I suspect most most devices hang on an I2C stop so a reset would probably would cause limited glitches since it may occur after all data is sent to the device.

greiman, 2021-04-29:
I replace AvrI2c.h with one that has time-outs and some other mods.  It shouldn't hang in a loop waiting for an I2C command to complete.

If there are glitches in the display we can pursue these.  Debug prints can be enabled in [AvrI2c.h](https://github.com/greiman/SSD1306Ascii/blob/master/src/utility/AvrI2c.h) at about line 78.

I did some experiments an found a hang was possible if the I2C bus didn't have pull-ups.

ZL1CVD, 2021-04-29:
Thank you. Stability issue resolved. I have not managed to lock up even once.
Now compiled code is bigger than Arduino IVR even with LTO enabled. This no longer such an issue for me as I reduced serial TX/RX buffer size to accommodate arduino wire code.

closed 2021-04-29 by ZL1CVD