greiman/SSD1306Ascii #80: No license = no legal use of the code by others

michpro, 2021-05-04:
Your code could be very useful, but would you please add some license to this repository (e.g. MIT, BSD or Apache or whatever you choose) as you may not be aware of it, but legally, without a license, no one can use your code. [http://opensource.stackexchange.com/q/1720/775](url)

d-a-v, 2021-05-04:
There is a license in the beginning of every source file

```
 * This Library is free software: you can redistribute it and/or modify
 * it under the terms of the GNU General Public License as published by
 * the Free Software Foundation, either version 3 of the License, or
 * (at your option) any later version.
 *
 * This Library is distributed in the hope that it will be useful,
 * but WITHOUT ANY WARRANTY; without even the implied warranty of
 * MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
 * GNU General Public License for more details.
 *
 * You should have received a copy of the GNU General Public License
 * along with the Arduino SSD1306Ascii Library.  If not, see
 * <http://www.gnu.org/licenses/>.
 ```

michpro, 2021-05-04:
Indeed, it is in the files, my omission, just a pity that it is a viral GNU GPL license, and not free like the MIT license, which makes the project useless for me :(

closed 2021-05-04 by michpro