greiman/SSD1306Ascii #83: Compile Error with Teensy 2.0

richteel, 2021-09-02:
I received the following error when compiling code for the Teensy 2.0 board.
"In file included from D:\Users\richt\OneDrive\Documents\Projects\PocketGPS\sketch_sep02a\sketch_sep02a.ino:5:0:
D:\Users\richt\OneDrive\Documents\Arduino\libraries\SSD1306Ascii\src/SSD1306Ascii.h: In member function 'bool SSD1306Ascii::tickerText(TickerState*, const String&)':
D:\Users\richt\OneDrive\Documents\Arduino\libraries\SSD1306Ascii\src/SSD1306Ascii.h:432:30: error: could not convert 'str' from 'const String' to 'bool'
     return tickerText(state, str ? str.c_str() : nullptr);
                              ^~~
Error compiling for board Teensy 2.0."

I had to change the function declaration on lines 431  to 433 from
  bool tickerText(TickerState* state, const String &str) {
    return tickerText(state, str ? str.c_str() : nullptr);
  }

to

  bool tickerText(TickerState* state, const String &str);

After changing the function declaration it compiled and worked fine.

greiman, 2021-09-03:
I can't use the above fix since that will break use of String on Arduino boards.  Teensy has a different version of String and gcc.

I will modify `bool tickerText(TickerState* state, const String &str);`  So it will work on Teensy and Arduino.

The function was submitted as a pull by a user and I failed to notice that the check on str is not needed since null can not be passed by reference. 

> The difference between pass-by-reference and pass-by-pointer is that pointers can be NULL or reassigned whereas references cannot. Use pass-by-pointer if NULL is a valid parameter value or if you want to reassign the pointer. Otherwise, use constant or non-constant references to pass arguments.

This is my planed change:
```
  bool tickerText(TickerState* state, const String &str) {
    return tickerText(state, str.c_str());
  }
```

richteel, 2021-09-03:
I just made that change and verified that it works on the Teensy.

Thank You

closed 2021-09-17 by richteel