greiman/SSD1306Ascii #85: Full Ascii character set - e.g.: Umlauts in Germany

modelrailroader, 2022-01-01:
Hello and A Happy New Year everybody, 

I tried different things, but had no luck. Is there a good possibility or software to create "GLCDFONTDECL" scripts with the full 255 character set for your library? Especially I need to use german umlauts in two different fonts. I tried GLCD Font Creator, but it didn't work. Can you help me?

greiman, 2022-01-01:
I did not create any fonts, just used publicly available fonts.  I don't know where you can get a "Extended ASCII" with the characters you want.  There are no standards for "Extended ASCII" and over 200 mappings.

When I wrote SSD1306Ascii I used fonts in the openGLCD format.

https://bitbucket.org/bperrybap/openglcd/wiki/Home

Many of the fonts I used are here:

https://bitbucket.org/bperrybap/openglcd/src/master/fonts/

Looks like the font creator is really old. See openGLCDFontCreator.zip here:

https://bitbucket.org/bperrybap/openglcd/downloads/

modelrailroader, 2022-01-02:
Thank you very much for your help. I created a new font with openGLCDFontCreator now and used 
`oled.write(uint8_t)`
to display the character. For using the font in a ticker as well, I replaced some ascii-characters and pasted in the needed german umlauts. Now everything is working. Thanks and all the best for 2022!

closed 2022-01-02 by modelrailroader

Kirmy34, 2022-03-19:
@modelrailroader Wich FontCreator did you use and how did you import it to the library?
When I export a Font from the GLCDFontCreator @greiman mentioned, I don't get any charachters printing on the screen.
Looking into the Font's file, it seems to be exported in a slightly different format... I cannot find the format spec. used in the library

modelrailroader, 2022-03-19:
Normally it's very easy to export fonts from the Font creator @greiman mentioned before. For importing into the library, you have to copy the created file into the fonts-folder. I'm expecting you did this already. How does your font file looks like? Could you post an example here?--Diese Nachricht wurde von meinem Android Mobiltelefon mit GMX Mail gesendet.

Kirmy34, 2022-03-19:
[TestFont.txt](https://github.com/greiman/SSD1306Ascii/files/8310020/TestFont.txt)

Thank you for taking the time.
If you put the TestFont side by side with a working Font some things don't quite add up.

- The First 2 Bytes are 0x00 -> size of zero indicates fixed width font, actual length is width * height
- Then there are seemingly 2 Bytes for each Width and Hight whereas in the working fonts there are only Byte for each

modelrailroader, 2022-03-19:
I know your problem! The Font creator you used is not that one that @greiman mentioned and that you have to use. You have to install the FontCreator from the links greiman posted above. That's another creator but not for the SSD1306ascii lib. If you have problems installing it, you have to install Java Runtime as well. If you have other problems, please tell me.

Kirmy34, 2022-03-19:
I could just about kiss you right now!
After a closer look at the other tool I was using it was exporting the font in "X-GLCD" format... I'm guessing X = eXtended...

Thank you so much for the help!

modelrailroader, 2022-03-19:
Great to hear that it's working. Just one additional note because I was struggling with it a long time: For <space> (ASCII Nr. 32) you have to write 2 into the width column. Without, you can't use space. Have fun programming!