greiman/SSD1306Ascii #87: oled.println(); not going in new line on display

microicRI, 2022-03-03:
Hi,
arduino IDE 1.8.19
ESP32 + OLED128x64

// Test for minimum program size.

#include <Wire.h>
#include "SSD1306Ascii.h"
#include "SSD1306AsciiWire.h"

// 0X3C+SA0 - 0x3C or 0x3D
#define I2C_ADDRESS 0x3C

// Define proper RST_PIN if required.
#define RST_PIN -1
char test[20];

SSD1306AsciiWire oled;
//------------------------------------------------------------------------------
void setup() {
  Wire.begin();
  Wire.setClock(400000L);

#if RST_PIN >= 0
  oled.begin(&Adafruit128x64, I2C_ADDRESS, RST_PIN);
#else // RST_PIN >= 0
  oled.begin(&Adafruit128x64, I2C_ADDRESS);
#endif // RST_PIN >= 0

  oled.setFont(Adafruit5x7);               //System5x7);
  oled.clear();
  oled.print(" Hello world");
  delay(2000);
  oled.clear();
  String test_str="";
  test_str=("damir");
  //oled.set2X();
  oled.print("duzina_test_str= "); int duzina_test_str=test_str.length();oled.println(duzina_test_str); oled.println();  // go to next line
 oled.write('\n');
  //oled.print("test_str= ");oled.println(test_str);
  delay(2000);
  test_str+="š"; //ŠđĐčČćĆžŽ");
  oled.println("duzina_test_str= ");duzina_test_str=test_str.length();oled.println(duzina_test_str);; oled.print("\n");//oled.print(", ");oled.println("test_str= ");oled.println(test_str);
  
  delay(5000);
/*
 for (int n=0;n<256;n++){
  oled.clear();
  int i=n;
  oled.print(i);
  oled.print("= ");
  oled.print((char)n);
  delay(500); 
  
 }*/
  
}
//------------------------------------------------------------------------------
void loop() {}``

greiman, 2022-03-03:
Might be that you are printing characters no in the font.  This will cause print to fial.

microicRI, 2022-03-05:
Thanks for your answer, but there is an example of code together with open
cause. Can you point me to program line where are you see that case you
mention. Once again thanks for your help.
Damir

greiman, 2022-03-06:
```
char str[] = "š";
void setup() {
  Serial.begin(9600);
  for (uint8_t i = 0; i < strlen(str); i++) {
    Serial.println((uint8_t) str[i], HEX);
  }
}
void loop() {
}
```

Output is the UFT8 two byte character:
```
C5
A1
```
SSD1306Ascii only supports ASCII characters - note the name has Ascii for a reason.