greiman/SSD1306Ascii #88: Is there a way to programmatically determine what Font is current?

JonRobert, 2022-03-14:
Thank you very much for this great library and for your support.  I've been using it for quite a while and absolutely love it.

My question:
I want to make a small macro (or function) to make the "oled.SSD1306Ascii::setCursor(0,0);" Row and column inputs more "natural".   I thought if I could determine the current font after calling  "oled.setFont(CalLite24); " it would be easy to implement.
If not its not a show stopper, I would just have to make the font a variable and use it to read   <font>_WIDTH  and <font>_HEIGHT

Thanks again for your work and for sharing.

Jon

greiman, 2022-03-14:
Not sure what you want.  These functions return properties of the current font:

```
//------------------------------------------------------------------------------
uint8_t SSD1306Ascii::fontHeight() const {
  return m_font ? m_magFactor*readFontByte(m_font + FONT_HEIGHT) : 0;
}
//------------------------------------------------------------------------------
uint8_t SSD1306Ascii::fontRows() const {
  return m_font ? m_magFactor*((readFontByte(m_font + FONT_HEIGHT) + 7)/8) : 0;
}
//------------------------------------------------------------------------------
uint16_t SSD1306Ascii::fontSize() const {
  return (readFontByte(m_font) << 8) | readFontByte(m_font + 1);
}
//------------------------------------------------------------------------------
uint8_t SSD1306Ascii::fontWidth() const {
  return m_font ? m_magFactor*readFontByte(m_font + FONT_WIDTH) : 0;
}
```
setCursor(x, y) is pretty much the standard way to set the cursor in a graphics library. It allows you to position to the finest address on the display.

greiman, 2022-03-14:
```
 /**
   * @return The current font pointer.
   */
  const uint8_t* font() const {return m_font;}
```

greiman, 2022-03-15:
I think these functions are documented in the[ html here](https://github.com/greiman/SSD1306Ascii/tree/master/doc).

Click on SD1306Ascii.html.  I will regenerate the html next time I update the library to make sure it matches the source.

JonRobert, 2022-03-15:
Hi Bill,

 

After you sent me the functions I read through the documentation again and found them.   I hadn’t used then in a while and was going on my old code that used was kind of clumsy.

 

Thank you so much for your follow up emails.

 

Regards

 

John

closed 2022-04-01 by JonRobert