greiman/SSD1306Ascii #89: Ticker Text demo question / maybe issue

JonRobert, 2022-04-01:
I'm using the example "TickerTextDemo.ino" with some minor modifications.
   1) changed to 128 x 32
   2) shortened message
   3) added some print lines for debug / understanding.
 
My goal is to understand enough to run Ticker test in the beginning on my program (perhaps in Setup) then move on to my program code.  I haven't figured out what to include in what I hoped could be a function that would Ticker the text then stop.

Issue I've noticed: (or maybe my misunderstanding)

`    uint8_t rtn = oled.tickerTick(&state);
    if (rtn <= RTN_CHECK) {     //  returns FALSE every loop except when 1st char of the message reaches the left edge, in this case its 16 in which case it is TRUE`

Can you tell me how to
* know when the message is fully off the left edge of the active area so I can stop the Ticker
* stop ticker and move on to other code?

BTW I think you've created and offered a great library.  I've used it often with good success.  

Thank you 

John

Full modified Code:

`// Simple Ticker demo for I2C 128x32 oled.

// The ticker shifts text through a one line field on the display.
// A queue of text pointers is maintained by the ticker. You must
// not modify a text string while its pointer is in the queue.
// The display queue size is define by TICKER_QUEUE_DIM in SSD1306Ascii.h.

// tickerTick() returns the current length of the display pointer queue.
// If tickerText() is called for a return <= one, the display is continuous.
// If tickerText() is called for return zero, all text will be shifted off
// the display before another string starts.
//

#include <Wire.h>
#include "SSD1306Ascii.h"
#include "SSD1306AsciiWire.h"

#define I2C_ADDRESS 0x3C

SSD1306AsciiWire oled;

// Ticker display
#define RTN_CHECK   1   // Try values of zero or one for RTN_CHECK.
#define numberLoops 3
#define tickerRate 50   // milliseconds
TickerState state;      // Ticker state. Maintains text pointer queue and current ticker state.
const char* text[] = {
  "Prog:  blah blah v0.02    "
};    // this is an array of pointers to the text.
uint32_t tickTime = 0;
int n = 0;
// Ticker display end

//------------------------------------------------------------------------------

void setup() {

  Serial.begin(115200);
  delay(1000);
  Wire.begin();
  Wire.setClock(400000L);

  oled.begin(&Adafruit128x32, I2C_ADDRESS);

  // Use <font>, field at row 2, mag2x, columns 16 through 100.
  oled.tickerInit(&state, ZevvPeep8x16, 2, false, 16, 100);

  oled.displayRemap (true);

  // Try this for full screen width with set1X.
  // oled.tickerInit(&state, Adafruit5x7, 2);
}

void loop() {

  if (tickTime <= millis()) {
    tickTime = millis() + tickerRate;

   // Should check for error. rtn < 0 indicates error.
    uint8_t rtn = oled.tickerTick(&state);

    if (rtn <= RTN_CHECK) {     // <<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<< returns false every loop except when 1st char of the message reaches the left edge, in this case its 16
      // Should check for error. Return of false indicates error.
      oled.tickerText(&state, text[(n++)%3]);
      Serial.println(".........................rtn test true");
    }
    else{
      Serial.println("rtn test false");
    }
  }

}
 // -- end loop --
`

greiman, 2022-04-01:
oled.tickerTick() returns how many strings are queued to be displayed.  You can add another string when the return value is less than TICKER_QUEUE_DIM by calling oled.tickerText().

When the return value is zero, all strings have been displayed.

greiman, 2022-04-01:
```
  /**
   * @brief Advance ticker by one pixel.
   *
   * @param[in,out] state Ticker state.
   * @return Number of entries in text pointer queue.
   */
  int8_t tickerTick(TickerState* state);
```
I should add that return of -1 indicates an error.

JonRobert, 2022-04-01:
> BG wrote: oled.tickerTick() returns how many strings are queued to be displayed. You can add another string when the return value is less than TICKER_QUEUE_DIM by calling oled.tickerText().
When the return value is zero, all strings have been displayed.

the below never printed.
>            Serial.println("                     >>>>>> finished displaying all strings");

**I never received a oled.tickerTick() of "0"**   Perhaps I'm doing something wrong or I misunderstood.

My Code:  ( I don't know why it doesn't display as code in the message, I've used the <> format)

`
// Simple Ticker demo for I2C 128x32 oled.
//
// oled.tickerTick() returns how many strings are queued to be displayed. You can add another string when the return value is less than TICKER_QUEUE_DIM by calling oled.tickerText().
//  When the return value is zero, all strings have been displayed.
//

#include <Wire.h>
#include "SSD1306Ascii.h"
#include "SSD1306AsciiWire.h"

#define I2C_ADDRESS 0x3C

SSD1306AsciiWire oled;

// Ticker display
#define RTN_CHECK   1   // Try values of zero or one for RTN_CHECK.
#define numberLoops 3
#define tickerRate 50   // milliseconds
TickerState state;      // Ticker state. Maintains text pointer queue and current ticker state.
const char* text[] = {
    "Prog:  blah blah v0.02    "   //  26 char  26 x (8 + 3) = 286, close to the 289 we counted in the number of "false" loops.
};    // this is an array of pointers to the text.
uint32_t tickTime = 0;
int n = 0;
// Ticker display end

//  Setup
void setup() {
    Serial.begin(115200);
    delay(1000);
    Wire.begin();
    Wire.setClock(400000L);
    oled.begin(&Adafruit128x32, I2C_ADDRESS);
    // Use <font>, field at row 2, mag2x, columns 16 through 100.
    oled.tickerInit(&state, ZevvPeep8x16, 2, false, 16, 100);
    oled.displayRemap (true);
    // Try this for full screen width with set1X.
    // oled.tickerInit(&state, Adafruit5x7, 2);
    oled.clear();
}

//----------------------------- Main

void loop() {
    if (tickTime <= millis()) {
        tickTime = millis() + tickerRate;

   // Should check for error. rtn < 0 indicates error, rtn of 0 indicates all strings have been displayed.
        uint8_t rtn = oled.tickerTick(&state);

        if (rtn == 0){
            Serial.println("                                         finished displaying all strings");
        }

        if (rtn <= RTN_CHECK) {     // rtb = false every loop except when 1st char of the message reaches the left edge
            // Should check for error. Return of false indicates error.
            oled.tickerText(&state, text[(n++)%3]);
            Serial.println(".........................rtn test true");
        }
        else{
            Serial.println("rtn test false  ");
            return;
        }
    }
}  // -- end loop --
//  -- eof --

`

greiman, 2022-04-01:
Yes you will never get return zero since you always queue another copy of the string here when rtn == 1.
```
 if (rtn <= RTN_CHECK) {     // rtb = false every loop except when 1st char of the message reaches the left edge
            // Should check for error. Return of false indicates error.
            oled.tickerText(&state, text[(n++)%3]);  <<------ Causes another copy of your string to be queued when rtn == 1.
            Serial.println(".........................rtn test true");
        } else{
            Serial.println("rtn test false  ");  <<----- This is not false it is more than one copy queued.
            return;
        }
```

JonRobert, 2022-04-02:
Got it :)  Thanks

closed 2022-04-02 by JonRobert