greiman/SSD1306Ascii #90: How to use u8g2_font_unifont_t_chinese Chinese character library

he77PC011, 2022-04-30:
/*
  Fontname: -gnu-Unifont-Medium-R-Normal-Sans-16-160-75-75-c-80-iso10646-1
  Copyright: Copyright (C) 1998-2019 Roman Czyborra, Paul Hardy,  Qianqian Fang, Andrew Miller, Johnnie Weaver, David Corbett, et al.  License GPLv2+: GNU GPL version 2 or later <http://gnu.org/licenses/gpl.html>  with the GNU Font Embedding Exception.
  Glyphs: 575/57086
  BBX Build Mode: 0
*/
const uint8_t u8g2_font_unifont_t_chinese2[20225] U8G2_FONT_SECTION("u8g2_font_unifont_t_chinese2") = 
  "\77\0\4\2\5\5\4\5\6\20\20\0\376\12\376\13\377\1\271\3x\5e \6\0\240G\1!\10A"
  "\61D\341 !\42\10\205(F!\343&#\20F%D\245'\203A\42\322b\60\210\364\4$\24G"
  "%D'\64PD\42\222\320l\24\221T\14D\31\0%\26G%DC#I\224$\42\212T,\225"

machmar, 2022-04-30:
In the name of the library, it says **SSD1306Ascii**. Ascii stands for _American Standard Code for Information Interchange_. It means that there are 255 defined symbols that can be displayed and unfortunately Chinese characters are not supported. Sorry.

greiman, 2022-04-30:
SSD1306Ascii supports mostly Standard 7-bit ASCII fonts with up to 96 graphic characters.

Adafruit5x7 supports an 8-bit extended ASCII set with 254 characters.

Some fonts like FIXEDNUMS15x31 only support 16 characters.