greiman/SSD1306Ascii #91: degrees celsius symbol

musapinar, 2022-06-25:
Hello,

How do I print code 167 of cp437? https://www.ascii-codes.com/
It refers to the "degrees Celsius" symbol.
I'm using System5x7.
I tried to include the cp437 header file from the fonts directory and then print it like ```oled.print(167)``` but no joy. It prints 167 as an integer. 

Thank you for your help.

greiman, 2022-06-26:
The degree symbol is the last character in the System5x7 font it is character 128 decimal or 200 octal.

This code:
```
  oled.setFont(System5x7);
  auto tempC = 25.0;
  auto tempF = tempC*9/5 + 32;
  oled.print(tempC, 1);  // Use one decimal digit
  oled.print(" \200C is ");  // \200 is octal escape for degree symbol
  oled.print(tempF, 1);
  oled.println(" \200F\n");
```
Should print:

> 25.0 °C is 77.0 °F

musapinar, 2022-06-26:
That did work. Thank you Bill.
I can see it's been documented since 2018 here : https://github.com/greiman/SSD1306Ascii/blob/master/src/fonts/System5x7.h#L144
It did not came up in the search results within the repo when I looked for "degrees". Sorry about that.
Have a great day.

greiman, 2022-06-26:
If you need symbols in cp437 you can use cp437font8x8 or Adafruit5x7.  For Adafruit5x7 you must change this symbol in Adafruit5x7.h. 

`#define ADAFRUIT_ASCII96 0  // was one`

To select the fonts use:
`  oled.setFont(cp437font8x8);`
  or
`  oled.setFont(Adafruit5x7);`

The down side is that println() will no longer work since \n and \r are now this bullet, ◙, and eighth note, ♪.

You must use cursor positioning and clear line instead of println(). 

 Also you must use escaped character codes.

closed 2022-07-10 by musapinar

greiman, 2025-03-22:
Also see[ this](https://github.com/greiman/SSD1306Ascii/issues/91).

mianqi2016, 2025-03-28:
Thanks,

` 
  oled.print(temp_val);
  oled.print(" ");
  oled.setFont(System5x7);
  oled.setCursor(85, 3);
  oled.print("\200C");
`

that is OK.