greiman/SSD1306Ascii #93: Software i2c

GigiG62, 2022-07-12:
Hi, is it possible to initiate the oled.begin on pins different from the hardware i2c pins?

greiman, 2022-07-12:
Looks like the default I2C pins are used internally on ESP32-Cam.  You can select alternate pins with `bool begin(int sda=-1, int scl=-1, uint32_t frequency=0);` .

Try changing this:
```
SSD1306AsciiWire oled;

void setup() {
  Wire.begin();
  Wire.setClock(400000L);
```
To:

```
SSD1306AsciiWire oled;

// select your pin choice here
#define I2C_SDA 14
#define I2C_SCL 15

void setup() {
  Wire.begin(I2C_SDA , I2C_SCL , 400000L);
```
The ESP32 has two I2C ports.

You can use the second port like this
```
SSD1306AsciiWire oled(Wire1);

// select your pin choice here
#define I2C_SDA 14
#define I2C_SCL 15

void setup() {
  Wire1.begin(I2C_SDA , I2C_SCL , 400000L);
```

I compiled the above but didn't test them on an ESP32.  Let me know if this works. Its been years since I used an alternate port.

GigiG62, 2022-07-16:
Is it possible to use a 128x128 OLED like [this](https://www.aliexpress.com/item/1005004168425279.html?spm=a2g0s.8937460.0.0.3da82e0eT67vdG)?

greiman, 2022-07-16:
No the SD1306 controller only supports up to 128x64.