greiman/SSD1306Ascii #94: 70x42 OLED ?

Sarah-C, 2022-07-23:
I've tried adding code for a 72x40 0.42" OLED display, like in the 01Space ESP32-C3..... https://github.com/01Space/ESP32-C3-0.42LCD

The new definition I added is this, but it causes corruption of pixels down the left side.

```
//------------------------------------------------------------------------------
/** Initialization commands for a 64x48 Micro OLED display (by r7) */
static const uint8_t MEM_TYPE MicroOLED72x40init[] = {
    // Init sequence for 64x48 Micro OLED module
    SSD1306_DISPLAYOFF,
    SSD1306_SETDISPLAYCLOCKDIV, 0x80,  // the suggested ratio 0x80
    SSD1306_SETMULTIPLEX, 0x2F,        //
    SSD1306_SETDISPLAYOFFSET, 0x0,     // no offset
    SSD1306_SETSTARTLINE | 0x0,        // line #0
    SSD1306_CHARGEPUMP, 0x14,          // internal vcc
    SSD1306_NORMALDISPLAY,
    SSD1306_DISPLAYALLON_RESUME,
    SSD1306_SEGREMAP | 0x1,            // column 127 mapped to SEG0
    SSD1306_COMSCANDEC,                // column scan direction reversed
    SSD1306_SETCOMPINS, 0x12,          // 0x12 if height > 32 else 0x02
    SSD1306_SETCONTRAST, 0x7F,         // contrast level 127
    SSD1306_SETPRECHARGE, 0xF1,        // pre-charge period (1, 15)
    SSD1306_SETVCOMDETECT, 0x40,       // vcomh regulator level
    SSD1306_DISPLAYON
};
/** Initialize a 64x48 Micro OLED display. */
static const DevType MEM_TYPE MicroOLED72x40 = {
  MicroOLED64x48init,
  sizeof(MicroOLED72x40init),
  72,
  40,
  0
};

```

greiman, 2022-07-23:
I look at [u8g2](https://github.com/olikraus/u8g2) for hints when there are problems with a display.

I looked at [u8x8_d_ssd1306_72x40.c](https://github.com/olikraus/u8g2/blob/master/csrc/u8x8_d_ssd1306_72x40.c).

The one thing I noticed for the 72x40 display was a default_x_offset of 28.  This is at about line 240.  

This probably is not the problem since you are using zero for column offset and only have a column of bad pixels.

You could try an offset of two since SH1106 controllers often require an offset of two.  There is a file [u8x8_d_sh1106_72x40.c](https://github.com/olikraus/u8g2/blob/master/csrc/u8x8_d_sh1106_72x40.cl) but it has an offset of 30.

May be you can see something else in these files.

Sorry I have no other ideas what is causing your problem.

Sarah-C, 2022-07-25:
Thanks for checking! I'll have a go after work.

Sarah-C, 2022-07-26:
@greiman hi!

Could you spare a moment to look at this sketch in the emulator?

Just press the green play button, and it fills the OLED with "123"'s.

When it gets to the bottom, is the screen behaving how you coded it? I thought it would scroll the text up a line, in which case I'll raise a bug with wokwi. If it behaves as it should, my Universal Terminal object will come into its own!  =)

https://wokwi.com/projects/338268461896565332

greiman, 2022-07-27:
The emulator does not behave correctly.  I write data in to the controller RAM as a circular buffer and map the display start line with a command.

I suspect that the emulator does not correctly implement this command:
```
/** Set display RAM display start line register from 0 - 63. */
#define SSD1306_SETSTARTLINE 0x40
```
It is used in this functions to map the RAM to the display:

```
void SSD1306Ascii::setStartLine(uint8_t line) {
  m_startLine = line & 0X3F;
  ssd1306WriteCmd(SSD1306_SETSTARTLINE | m_startLine);
}
```

Sarah-C, 2022-07-31:
I'll let the authors know. Thank you!

Sarah-C, 2022-07-31:
As it's unrelated to this library, I'm closing the issue. Thank you all.

closed 2022-07-31 by Sarah-C

urish, 2022-07-31:
@Sarah-C thanks for reporting! I ran your code against physical hardware, and can see it indeed behaves differently:

https://user-images.githubusercontent.com/892318/182016020-de7aff16-abe9-489c-a9b9-df243b94856c.mp4

I opened an issue to track the bug on wokwi/wokwi-features#394

urish, 2022-07-31:
Update: the simulator issue was fixed!