greiman/SSD1306Ascii #95: Begin() return error is OLED not connected.

QuidneIT, 2022-10-31:
The Adafruit OLED library returns an error if the OLED is not present.
e.g. if(!display.begin(SSD1306_SWITCHCAPVCC, 0x3C))

Is there a way this can be done with this library?

greiman, 2022-10-31:
begin() is void so no error is returned.

You could simply check to see if a device responds to the I2C address after the Wire.begin() call.

This how a typical I2C scanned checks for devices.

```
    Wire.beginTransmission(address);
    byte error = Wire.endTransmission();

    if (error == 0) {
      Serial.print("I2C device found at address 0x");
      if (address < 16) {
        Serial.print("0");
      }
      Serial.print(address, HEX);
      Serial.println("  !");
```

Several other users do this when a display may not be present.

QuidneIT, 2022-10-31:
Thank you. Was thinking of this solution as well if the library did not have the build in functionality.

closed 2022-10-31 by QuidneIT