greiman/SSD1306Ascii #96: Text height

yusufpraditya, 2022-12-21:
How to get text height just like text width?
I can get the text width by using this code
`int textWidth = oled.strWidth("Test");`

yusufpraditya, 2022-12-21:
after reading the documentation, its impossible.. i just want to center  the text horizontally and vertically.. i can center the text horizontally using calculation of text width, but i still have no idea to center the text vertically

greiman, 2022-12-21:
You can only choose the row and rows are separated by eight pixels.  It is a hardware property.

fontHeight() gives the height in pixels but you can not truly center a line vertically since the display hardware is only row, eight pixel, addressable.

closed 2023-01-06 by yusufpraditya