greiman/SSD1306Ascii #98: Compiling error with Arduino nano every (ATMega4809)

francoispap, 2023-02-01:
Hello,
I am successfully using your code with a regular Arduino Nano Board
Due to the need of more code space, I have tried to merge my code to a nano Every
I am getting the following error "AvrI2c not supported for this chip"

Are you thinking that some changes in the your library code  will fix this issue

If it is a big work on your side to fix that,  what more enhanced Arduino board would be compatible with your library ?

Thanks for your time if you can have a look

Regards
Pierre

greiman, 2023-02-01:
I will not be adding custom I2C for any chips in the future.  There are too many AVR variants and "almost" clones that are not compatible with the original AVR I2C controller. 

Any board supported by the Arduino IDE with a Wire library should work.

francoispap, 2023-02-01:
Thanks for your feedback and explanations
Regards
Pierre

closed 2023-08-19 by francoispap

francoispap, 2023-08-19:
I have moved my project to nano ESP32 board.
Greiman code works perfectly with using Software SPI communication (SSD1306-/1309) Library "SSD1306AsciiSoftSpi.h"
I2C doesn't work