greiman/SSD1306Ascii #104: When using a display with more than 64 pixels in the Y direction, the lcdHeight parameter is ignored.

jonathanmlang, 2023-03-01:
In SSD1306Ascii.cpp there is an issue that prevents displays with more than 8 rows or 64 pixels from displaying more than that.
The issue seems to be at line 166 in that file.

I have changed:
`ssd1306WriteCmd(SSD1306_SETSTARTPAGE | ((m_row + m_pageOffset) & 7);`

to:
`ssd1306WriteCmd(SSD1306_SETSTARTPAGE | ((m_row + m_pageOffset) & ((m_displayHeight/8)-1)) );`

This seems to have solved the problem in my case, using a vertically orientated SH1107 oled with a resolution of 64x128.
I hope this helps!

greiman, 2023-03-01:
I will Add this.

Also  on horizontal bar graphs, you can use:

`void ssd1306WriteRamBuf(uint8_t c);`

jonathanmlang, 2023-03-01:
Thanks for that ill give it a try. This is by far my favourite display library so thanks for your work. Just trying to contribute a bit to show my appreciation.

greiman, 2023-03-01:
Your SH1107 mod may not work.  The reason for the mask of 7 was not the display height but the memory pages in the SSD1306.

The SSD1306 has a 128x64 memory layout.

The SH1107 supports 128x128  so must have a different memory page layout.

jonathanmlang, 2023-03-01:
Indeed, could be all sorts ive missed.

greiman, 2023-03-01:
I tried the horizontal bar graph like this for 50 pixels long.  I added this loop th Hello World:
```
  oled.print("Hello world!");
  // Test bar graph
  oled.println();
  for (uint8_t i = 0; i < 50; i++) {
    oled.ssd1306WriteRamBuf(0b00111100);
  }
```

jonathanmlang, 2023-03-02:
Yes I like that, using that function I have achieved a vertical bargraph too.