greiman/SSD1306Ascii #110: Some functions seem to need pixels as a parameter and not row/column numbers?

steve6375, 2023-06-01:
First many thanks for this code!!! It is really lightweight which was what I needed to get an SD card working with a display because the AdafruitGFX library uses up all resources of the poor little Arduino Uno!

I have been playing with some code and find that the documentation does not seem to match my experiments.
The doc seems to imply that some commands need a row or column number, but I can only get it to work by using pixel number in some commands.

Please see code below (which appears to work) and especially these lines
```
oled.clearField((fontwidth+1)*3,2,4);   // clear 2nd line row 3 for 4 chars (0-based line and row???) 4th char ..7th char are deleted
oled.setCursor((fontwidth+1)*13,1) ;    // col, row  - set cursor on 2nd line char 13 
oled.clearToEOL();                      // clear rest of line from current cursor pos, cursor does not move
oled.clear((fontwidth+1)*9,((fontwidth+1)*11)-1,1,3);  // col,col,row,row   rows 0,1,2  delete 4th column
oled.setStartLine((fontheight+1)*1);    // display starting at pixel line x (typ, 0, 8, 16 or 24, etc.)
oled.scrollDisplay((fontheight+1)*1);   // scroll 1-255 pixel lines
```

Am I doing something wrong? Some parameters seem to need pixel values and others character row/column  values?

[image]
[image]

```
void setup() {
  Serial.begin(115200);
  Serial.println("Start");
  Serial.println("====================== Start ======================");
  // Use next line if no RST_PIN or reset is not required.
  // oled.begin(&Adafruit128x32, CS_PIN, DC_PIN, OLED_CLK, OLED_DATA);  
  oled.begin(&Adafruit128x32, OLED_CS, OLED_DC, OLED_CLK, OLED_MOSI, OLED_RESET);
  oled.setFont(Adafruit5x7);  
  oled.setFont(lcd5x7);  
  oled.setFont(Wendy3x5);  
  oled.setFont(Iain5x7);
  oled.setFont(Stang5x7);
  oled.setFont(System5x7);
  oled.setFont(SystemFont5x7);
  // Set auto scrolling at end of window.
  //  SCROLL_MODE_OFF  - newline will not scroll the display or RAM window.
  //  SCROLL_MODE_AUTO - newline will scroll both the display and RAM windows.
  //  SCROLL_MODE_APP  - newline scrolls the RAM window. The app scrolls the display window.
  oled.setScrollMode(SCROLL_MODE_AUTO);
oled.setContrast(80);  // 0 - 255
uint32_t fontwidth = oled.fontWidth();  // e.g. 5 for 5x7 font
Serial.print("fontwidth-");
Serial.println(fontwidth);
uint32_t fontheight = oled.fontHeight();  // e.g. 7 for 5x7 font
Serial.print("fontHeight-");
Serial.println(fontheight);
oled.clear(); 
oled.println("123456789012345678901234");  // max string before runs off end 21.3 chars for 5x7 font  128x32 display
oled.print("223456789012345678901234");
oled.println(F("\n323456789012345678901234"));
oled.print("423456789012345678901234");
oled.setInvertMode(1);
oled.print("\n523456789012345678901234");
oled.print("\n623456789012345678901234");
oled.print("\n723456789012345678901234");
oled.setInvertMode(0);
oled.setContrast(255);  // 0 - 255
oled.setCursor((fontwidth+1)*13,2) ;    // col, row  - starts at 0-based
oled.print("X");
oled.write(0x55);  // write ASCII char U at current location

oled.clearField((fontwidth+1)*3,2,4);   // clear 2nd line row 3 for 4 chars (0-based line and row???) 4th char ..7th char are deleted
delay(1000);
oled.setCursor((fontwidth+1)*13,1) ;    // col, row  - set cursor on 2nd line char 13 
oled.clearToEOL();                      // clear rest of line from current cursor pos, cursor does not move
delay(1000);
oled.clear((fontwidth+1)*9,((fontwidth+1)*11)-1,1,3);  // col,col,row,row   rows 0,1,2  delete 4th column

// scroll test
delay(10000);
// scroll up to line 2 at top
oled.setStartLine((fontheight+1)*1);    // display starting at pixel line x (typ, 0, 8, 16 or 24, etc.)
delay(3000);
// scroll down one line to line 3  at top
oled.scrollDisplay((fontheight+1)*1);   // scroll 1-255 pixel lines
delay(5000);

}
```

greiman, 2023-06-02:
Columns are addressed by pixel and rows are addressed by 8 pixels. The reason is that a single byte maps to eight vertical pixels.

The SSD1306 Memory is write only for I2C and SPI so the only way to implement a library without an buffer of the display in host memory is to use this hardware feature.

The only exception is the mapping of SSD1306  memory to the OLED display is by pixel address.  This allows efficient scrolling.

The datasheet uses (column, segment) for locations on the display.  Most SSD1306 libraries refer to display memory with row (page number) for vertical position and column used for horizontal position.

[image]

steve6375, 2023-06-02:
Ok, but the documentation does not make this clear at all, it says row and column with no distinction in units?

greiman, 2023-06-02:
I defined the terms in a few place like here but not for every function.  Here is the key place in the documentation where height in pixels verses rows matters.  Fonts height is in pixels.  Vertical address is in 8-pixel units.  Horizontal address and widths is always in pixels.  Functions with row arguments are in units of 8-pixels. 

Here is a place where that is defined.

[image]

One other measure is scrolling is in terms of lines.  Lines are the height of the current font in rows.

So height can be in pixels for characters in a font, rows for addresses or lines for scrolling.