greiman/SSD1306Ascii #114: How to make 3 lines of text as large as possible

tobivan, 2023-10-29:
Hi there,

first: thanks a lot for this great library which saves a lot of space, too.

Using a 128x64 0.96" SSD1306 OLED, I wonder how I could have 3 lines with text as large as possible. If I set "oled.set2X();" there are only 2 full lines possible, as far as I can see. If I instead set a larger font e.g. "Cooper26" there is a lot of space between the lines so there can only be a little more than 2 lines on the display.

Any idea on how to do that?
Thanks a lot!
Tobias

greiman, 2023-10-29:
The basic hardware has eight rows that are each  8-bits high. SSD1306Ascii doe not use an internal bit map buffer so lines must be a multiple of row height.

Your only choices are eight lines with one row each, four lines with two rows per line or two line with either three or four rows per line.

You could use another library with an internal 1 KB buffer.  Here are two popular libraries:

https://github.com/adafruit/Adafruit_SSD1306

https://github.com/olikraus/u8g2

tobivan, 2023-10-29:
I see, thanks a lot for your answer.
Best regards