greiman/SSD1306Ascii #117: When using the ssd1306AvrI2c library, the avr hangs indefinitely

jonathanmlang, 2023-12-19:
This happens when noise occurs on the i2c lines or the display is disconnected. Only way out is to reset. This problem can be dealt with in the wire library variant using wire.setWireTimeout(). But there doesnt seem to be a version to use with your avri2c library. I prefer using it as well because its so much faster than wire and would like to get it working well. Thanks in advance.

greiman, 2023-12-19:
There is a timeout loop in AvrI2c  but it takes too long.

The loop count is [here](https://github.com/greiman/SSD1306Ascii/blob/c43eaa9db195d17277ffc3d1ec11e3b21ba37775/src/utility/AvrI2c.h#L238).

I tried changing the 0XFFFF to 0XFF and examples using AvrI2c still seems to work.  

The AvrI2c128x64 example runs in about 341 ms with the shorter timeout if you remove the display.  It runs in 71 ms with the display.

A major problem is lack of status returns form member functions.  I just don't have time now to add functionality to my Arduino libraries.

jonathanmlang, 2024-02-18:
Ive just tried this, the program no longer hangs but the display continues to remain frozen. Is there a way of triggering a reinitialisation routine for the display when this occurs? like running oled.end() then oled.begin()?

greiman, 2024-02-18:
I can't help much since I don't have the hardware here with the problem.

Some I2C displays have a reset pin.  I have not been able to find a software reset command,  Some posts on the internet claim the command 0XE4 does a reset for their display. 

There are lots of posts on the internet looking for a solution to reset for I2C displays.

If the I2C controller is hung, it might be possible to reset it with an call to the end() function [here](https://github.com/greiman/SSD1306Ascii/blob/c43eaa9db195d17277ffc3d1ec11e3b21ba37775/src/utility/AvrI2c.h#L123).

I don't have an end() function in SSD1306Ascii since I don't know what it should do since there is no reset command.

what happens if you call oled.begin() when it hangs.

jonathanmlang, 2024-02-19:
Well my display has no reset pin however resetting the arduino board does bring the display back to life.